libimobiledevice/libimobiledevice #413: GnuTLS fails to connect to iOS10 mobile devices via libimobiledevice

samrocketman, 2016-11-03:
# Problem

`libimobiledevice` and `ifuse` can't connect to iOS10 devices when linked against GnuTLS (all built from the latest master HEADs). However, it works when libimobiledevice is linked with OpenSSL.

See https://gitlab.com/gnutls/gnutls/issues/145 where the GnuTLS maintainer suggested I open an issue with libimobiledevice.  Nikos, the maintainer, was nice enough to suggest a few code edits which I applied with `patch`.  However, `libimobiledevice` still can't connect to iOS 10 devices when linked with GnuTLS.

# Without the patch

```
$ export GNUTLS_DEBUG_LEVEL=99

$ idevicepair pair
gnutls[2]: Enabled GnuTLS 3.5.5 logging...
gnutls[2]: getrandom random generator was detected
gnutls[2]: Intel SSSE3 was detected
gnutls[2]: Intel AES accelerator was detected
gnutls[2]: Intel GCM accelerator (AVX) was detected
gnutls[2]: unable to access: /etc/gnutls/default-priorities: 2
gnutls[3]: ASSERT: common.c[_gnutls_x509_export_int_named]:555
gnutls[3]: ASSERT: common.c[_gnutls_x509_export_int_named]:555
gnutls[3]: ASSERT: common.c[_gnutls_x509_export_int_named]:555
gnutls[3]: ASSERT: common.c[_gnutls_x509_export_int_named]:555
gnutls[3]: ASSERT: pk.c[wrap_nettle_pk_fixup]:1818
gnutls[3]: ASSERT: privkey.c[gnutls_x509_privkey_import_rsa_raw2]:979
SUCCESS: Paired with device 37b633350ab83dc815a6a97dcd6d327b12c41968

$ ifuse ~/usr/mnt/
gnutls[2]: Enabled GnuTLS 3.5.5 logging...
gnutls[2]: getrandom random generator was detected
gnutls[2]: Intel SSSE3 was detected
gnutls[2]: Intel AES accelerator was detected
gnutls[2]: Intel GCM accelerator (AVX) was detected
gnutls[2]: unable to access: /etc/gnutls/default-priorities: 2
gnutls[5]: REC[0x2060fc0]: Allocating epoch #0
gnutls[3]: ASSERT: constate.c[_gnutls_epoch_get]:600
gnutls[5]: REC[0x2060fc0]: Allocating epoch #1
gnutls[4]: HSK[0x2060fc0]: Keeping ciphersuite: GNUTLS_RSA_AES_128_CBC_SHA1 (00.2F)
gnutls[4]: HSK[0x2060fc0]: Keeping ciphersuite: GNUTLS_RSA_AES_256_CBC_SHA1 (00.35)
gnutls[4]: EXT[0x2060fc0]: Sending extension Extended Master Secret (0 bytes)
gnutls[4]: EXT[0x2060fc0]: Sending extension Encrypt-then-MAC (0 bytes)
gnutls[4]: EXT[0x2060fc0]: Sending extension OCSP Status Request (5 bytes)
gnutls[4]: EXT[0x2060fc0]: Sending extension Safe Renegotiation (1 bytes)
gnutls[4]: EXT[0x2060fc0]: Sending extension Session Ticket (0 bytes)
gnutls[4]: HSK[0x2060fc0]: CLIENT HELLO was queued [75 bytes]
gnutls[11]: HWRITE: enqueued [CLIENT HELLO] 75. Total 75 bytes.
gnutls[11]: HWRITE FLUSH: 75 bytes in buffer.
gnutls[5]: REC[0x2060fc0]: Preparing Packet Handshake(22) with length: 75 and min pad: 0
gnutls[9]: ENC[0x2060fc0]: cipher: NULL, MAC: MAC-NULL, Epoch: 0
gnutls[11]: WRITE: enqueued 80 bytes for 0x20648f0. Total 80 bytes.
gnutls[5]: REC[0x2060fc0]: Sent Packet[1] Handshake(22) in epoch 0 and length: 80
gnutls[11]: HWRITE: wrote 1 bytes, 0 bytes left.
gnutls[11]: WRITE FLUSH: 80 bytes in buffer.
gnutls[11]: WRITE: wrote 80 bytes, 0 bytes left.
gnutls[3]: ASSERT: buffers.c[get_last_packet]:1159
gnutls[10]: READ: -2 returned from 0x20648f0, errno=0 gerrno=0
gnutls[3]: ASSERT: buffers.c[_gnutls_stream_read]:367
gnutls[3]: ASSERT: buffers.c[_gnutls_io_read_buffered]:587
gnutls[3]: ASSERT: record.c[recv_headers]:1055
gnutls[3]: ASSERT: record.c[_gnutls_recv_in_buffers]:1175
gnutls[3]: ASSERT: buffers.c[_gnutls_handshake_io_recv_int]:1414
gnutls[3]: ASSERT: handshake.c[_gnutls_recv_handshake]:1459
gnutls[3]: ASSERT: handshake.c[handshake_client]:2822
gnutls[5]: REC[0x2060fc0]: Start of epoch cleanup
gnutls[5]: REC[0x2060fc0]: End of epoch cleanup
gnutls[5]: REC[0x2060fc0]: Epoch #0 freed
gnutls[5]: REC[0x2060fc0]: Epoch #1 freed
GnuTLS error: Error in the pull function.
Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.
```
[... 70 lines cut ...]
gnutls[9]: ENC[0x207f230]: cipher: NULL, MAC: MAC-NULL, Epoch: 0
gnutls[11]: WRITE: enqueued 80 bytes for 0x207c610. Total 80 bytes.
gnutls[5]: REC[0x207f230]: Sent Packet[1] Handshake(22) in epoch 0 and length: 80
gnutls[11]: HWRITE: wrote 1 bytes, 0 bytes left.
gnutls[11]: WRITE FLUSH: 80 bytes in buffer.
gnutls[11]: WRITE: wrote 80 bytes, 0 bytes left.
gnutls[3]: ASSERT: buffers.c[get_last_packet]:1159
gnutls[10]: READ: -2 returned from 0x207c610, errno=0 gerrno=0
gnutls[3]: ASSERT: buffers.c[_gnutls_stream_read]:367
gnutls[3]: ASSERT: buffers.c[_gnutls_io_read_buffered]:587
gnutls[3]: ASSERT: record.c[recv_headers]:1055
gnutls[3]: ASSERT: record.c[_gnutls_recv_in_buffers]:1175
gnutls[3]: ASSERT: buffers.c[_gnutls_handshake_io_recv_int]:1414
gnutls[3]: ASSERT: handshake.c[_gnutls_recv_handshake]:1459
gnutls[3]: ASSERT: handshake.c[handshake_client]:2822
gnutls[5]: REC[0x207f230]: Start of epoch cleanup
gnutls[5]: REC[0x207f230]: End of epoch cleanup
gnutls[5]: REC[0x207f230]: Epoch #0 freed
gnutls[5]: REC[0x207f230]: Epoch #1 freed
GnuTLS error: Error in the pull function.
Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.
```

I'm not able to successfully connect to iOS 10 with any combination of code changes.  I'm not personally capable of fixing this myself however I feel libimobiledevice continuing to support GnuTLS as an option is important.

samrocketman, 2016-11-03:
This relates to https://github.com/libimobiledevice/libimobiledevice/issues/346 but is not a duplicate.

nikias, 2016-11-04:
The pairing record generation with newer GnuTLS was broken. I just fixed it with commit https://github.com/libimobiledevice/libimobiledevice/commit/23069d10341ce637fdad7321d447c53752dba48c.
Make sure to remove invalid paring records by deleting the appropriate files or call ```idevicepair unpair``` for the affected devices.

samrocketman, 2016-11-04:
Thanks, I'll test it and report back.

samrocketman, 2016-11-04:
I tested it and it works.  Thanks for the fix!

closed 2016-11-04 by samrocketman