libimobiledevice/libimobiledevice #414: Is it possible to skip the setup wizard pages after restoring device settings using idevicebackup2?

crummy, 2016-11-03:
1. Run `idevicebackup2 backup /tmp/iphonebackup`
2. Run `idevicebackup2 restore /tmp/iphonebackup --system --reboot`
3. Wait for reboot

Behaviour: After a reboot, "Update Complete", then "Welcome to iPhone" screens are present. These interfere with automated operation of the device (e.g. launching apps, running XCUItests).

Is it possible to skip this setup wizard after restoring settings? I'd love to use this feature to clean devices in a device farm but that won't work if I have to manually interact with them afterwards.

mexmer, 2016-11-03:
i don't think it's possible. even if you do restore trough iTunes, it always shows this screen. And it started doing that since iOS 9 i believe.

crummy, 2017-04-04:
I thought I had some luck with `cfgutil prepare --skip-all`, described as "Skip all possible panes of Setup Assistant", but it still didn't skip those. A real shame.

closed 2017-04-04 by crummy

mexmer, 2017-04-04:
sounds like some apple "improvement", if you can't skip it event using their configuration tool

wasifalbert, 2017-10-10:
Can anyone tell me where i can download cfgutil for window?

SGprooo, 2019-08-21:
I recall seeing a flag on purplebuddy.plist in settings that is “ForceNoBuddy”, and setup assistant will volunteer to exit if the flag is in place.

tausift0, 2020-04-24:
It is possible to skip setup in softwares like 3uTools and iMazing. How are those apps able to do it? Is there any way we can do that in libi?

SGprooo, 2020-04-24:
Far as I’m aware, 3u and probably all other similar software are just a GUI for libi.
These bypassing methods may be in code somewhere; I’d like to know how it’s done as well.

truonggiang0710, 2020-05-25:
Hi @SGprooo ,
Do you have any idea for that ?
I don't think iMazing tool use restore/Backup service. Because I don't see anything like restore process during exit setup assistant.

SGprooo, 2020-05-25:
> Hi @SGprooo ,
> 
> Do you have any idea for that ?
> 
> I don't think iMazing tool use restore/Backup service. Because I don't see anything like restore process during exit setup assistant.
> 
> 

3u works just fine, it’s literally free and gives you the option to bypass during setup.

truonggiang0710, 2020-05-27:
Hi @SGprooo ,
Sure, iMazing or 3utool can do it. But I want to know what service is used here.
Remove Setup.app&killall -9 springboard without jailbroken iPhone OR using any service like as com.apple.purplebuddy. The com.apple.purplebuddy.plist file after running exit setup assistant by iMazing.
`?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>ForceNoBuddy</key>
	<true/>
	<key>GuessedCountry</key>
	<dict>
		<key>at</key>
		<date>2020-05-25T03:07:06Z</date>
		<key>countries</key>
		<array>
			<string>US</string>
		</array>
	</dict>
	<key>HSA2UpgradeMiniBuddy3Ran</key>
	<true/>
	<key>LockdownSetLocale</key>
	<true/>
	<key>PaymentMiniBuddy4Ran</key>
	<true/>
	<key>SetupDone</key>
	<true/>
	<key>SetupFinishedAllSteps</key>
	<true/>
	<key>SetupVersion</key>
	<integer>11</integer>
	<key>WebKitAcceleratedDrawingEnabled</key>
	<false/>
	<key>chronicle</key>
	<dict>
		<key>features</key>
		<array/>
	</dict>
	<key>lastPrepareLaunchSentinel</key>
	<array>
		<date>2020-05-25T04:14:49Z</date>
		<integer>0</integer>
	</array>
	<key>setupMigratorVersion</key>
	<integer>10</integer>
</dict>
</plist>`
Do you have any idea ?

a-gunter, 2021-01-13:
Continuing this conversation in #934 if anybody is still looking into this / has discovered anything

SaadAnsari-sankey, 2021-12-21:
> 1. Run `idevicebackup2 backup /tmp/iphonebackup`
> 2. Run `idevicebackup2 restore /tmp/iphonebackup --system --reboot`
> 3. Wait for reboot
>
> Behaviour: After a reboot, "Update Complete", then "Welcome [...]

you find any solutions for this

crummy, 2021-12-21:
No I did not. This was years ago at Sauce Labs. Try  someone there at their real device team to see if they've made further progress.