libimobiledevice/libimobiledevice #415: Issues with deleting on Windows

schuylerdaddy, 2016-11-07:
I have encountered two problems with deleting files and folders on Windows and solved them both:
1)The calls to 'DeleteFile' and 'RemoveDirectory' in winbaseapi will fail with standard POSIX paths using forward slashes.  They file paths must have backslashes.  I will append a patch at the bottom that fixes that

2)Both 'DeleteFile' and 'RemoveDirectory' are actually macros that either resolve to [DeleteFileW,RemoveDirectoryW] for unicode or [DeleteFileA,RemoveDirectoryA] for UTF-8.  By default, VS2015 projects are unicode and those calls will fail.  Users would need to configure their project properties to use multi-byte instead of unicode to have those macros expand correctly.  Although, if the filenames are guaranteed to be utf8, I propose using  [DeleteFileA,RemoveDirectoryA] instead of  [DeleteFile,RemoveDirectory] altogether.

Patch:

diff --git a/tools/idevicebackup2.c b/tools/idevicebackup2.c
index bb7f5c3..60022c1 100644
--- a/tools/idevicebackup2.c
+++ b/tools/idevicebackup2.c
@@ -69,6 +69,17 @@ static int quit_flag = 0;
 
 #define PRINT_VERBOSE(min_level, ...) if (verbose >= min_level) { printf(__VA_ARGS__); };
 
+#ifdef _MSC_VER
+void replace_forward_slashes(const char* path) {
+	char* iter = path;
+	while (*iter) {
+		if (*iter == '/')
+			*iter = '\\';
+		++iter;
+	}
+}
+#endif
+
 enum cmd_mode {
 	CMD_BACKUP,
 	CMD_RESTORE,
@@ -212,6 +223,7 @@ static int remove_file(const char* path)
 {
 	int e = 0;
 #ifdef WIN32
+	replace_forward_slashes(path);
 	if (!DeleteFile(path)) {
 		e = win32err_to_errno(GetLastError());
 	}
@@ -240,6 +252,7 @@ static int remove_directory(const char* path)
 
 static int rmdir_recursive(const char* path)
 {
+	replace_forward_slashes(path);
 	DIR* cur_dir = opendir(path);
 	if (cur_dir) {
 		struct dirent* ep;

samrocketman, 2016-11-08:
Your patch is not legible.  Please wrap it in [code blocks](https://help.github.com/articles/basic-writing-and-formatting-syntax/#quoting-code) or submit a pull request.

mexmer, 2016-11-08:
this seems to me quite weird, since i'm using paths with forward slashes all time in windows, i suspect you have different problem there.

WINAPI calls work either with forward or backward slashes, you can even call following and it will work ::DeleteFile(_T("C:\\Temp\\Some/Other/Folder/With/File.tmp" ));

WINAPI internally converts path separators. see note here
https://msdn.microsoft.com/en-us/library/windows/desktop/aa365247(v=vs.85).aspx

> Note:  File I/O functions in the Windows API convert "/" to "\" as part of converting the name to an NT-style name, except when using the "\\?\" prefix as detailed in the following sections.

schuylerdaddy, 2016-11-08:
@mexmer thanks for the feedback.  I struggled with this earlier and was certain that was part of the issue.  I tried to reproduce the issue without replacing the forward slashes and was unable to do so.  I don't feel the first issue and the patch are necessary, again thanks for your time on that.

 I didn't need to use _T because I set the project settings to MultiByte.  That still leaves the following question.  Are the c-strings passed to remove_file and remove_directory always UTF8?   Regardless I feel the _T should be added to DeleteFile and RemoveDirectory to ensure it works correctly no matter what the project settings are in VS.

@samrocketman thank you for looking at this, I retract my patch as it is unnecessary.

mexmer, 2016-11-09:
Using _T macro is matter of habit, if you unsure if code will be compiled for unicode or or ansi. Nowadays people do not use ansi.

Same thing as using TCHAR and not char/wchar_t

schuylerdaddy, 2016-11-11:
@mexmer right, and the source here in the ifdef Wins2 lacks the _T.  I was saying maybe it should have it.

Anyhow I appreciate the responses and the time.  I'm going to close this issue, because there is no immediate issue to resolve.  Thank you.

closed 2016-11-11 by schuylerdaddy