libimobiledevice/libimobiledevice #416: simple (should be anyway): where can my iOS write a log file to where it can retrieved using libimobiledevice?

jpmhouston, 2016-11-09:
I'm working on an iOS app, reaching the end of my rope for official ways to export a log file from my app. I'm installing app by build & run in Xcode, from .ipa's, using app in the field, returning them and connecting to a developer Mac, so nothing fancy needed like streaming logs to a service or capture from remote users or anything. Between Xcode & iOS being so in flux and flaky right now, nothing is working.

Would appreciate if anyone could tell me where my iOS app should write a log file to so it can be accessed later using this project. Or if this is out of scope for this project, my apologies.

Alternately if there's a open source component (cocoapod etc) that implements something to make this kind of thing easy, adding maybe in-app functionality for displaying and exporting captured logs. This wouldn't be too hard to make myself but I don't need to push my schedule by another X days to solve what should be an already-solved problem.

nikias, 2016-11-09:
Enable File Sharing in your app's info.plist via Xcode, then access the Documents folder via HouseArrest.

jpmhouston, 2016-11-09:
To anyone finding this thread later, also see: http://stackoverflow.com/questions/4546334/uifilesharingenabled-has-no-effect

closed 2016-11-09 by jpmhouston

aymenim, 2016-11-14:
That use to be the case but after iOS 8 It has an effect only apps with that flag set could be accessed.