libimobiledevice/libimobiledevice #421: *** Error in `idevicebackup2': free(): invalid pointer: 0x00007f1840bb1288 ***

c0d3z3r0, 2016-11-22:
$ idevicebackup2 -i backup Stuff/iOSBackup/
Backup directory is "Stuff/iOSBackup/"
Started "com.apple.mobilebackup2" service on port 53072.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Backup will be encrypted.
Requesting backup from device...
Incremental backup mode.
Sending 'XXXXX/Status.plist' (189 Bytes)
Sending 'XXXXX/Manifest.plist' (39.3 KB)
Sending 'XXXXX/Manifest.db' (9.2 MB)
Sending 'XXXXX/Status.plist' (189 Bytes)
*** Error in `idevicebackup2': free(): invalid pointer: 0x00007f1840bb1288 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x70c4b)[0x7f1840888c4b]
/usr/lib/libc.so.6(+0x76fe6)[0x7f184088efe6]
/usr/lib/libc.so.6(+0x777de)[0x7f184088f7de]
/usr/lib/libplist.so.3(+0x2452)[0x7f1840dd5452]
/usr/lib/libplist.so.3(+0x89e8)[0x7f1840ddb9e8]
/usr/lib/libplist.so.3(+0x8a1f)[0x7f1840ddba1f]
idevicebackup2[0x405539]
/usr/lib/libc.so.6(__libc_start_main+0xf1)[0x7f1840838291]
idevicebackup2[0x406c1a]
======= Memory map: ========
00400000-0040d000 r-xp 00000000 fe:01 10512978                           /usr/bin/idevicebackup2
0060c000-0060d000 r--p 0000c000 fe:01 10512978                           /usr/bin/idevicebackup2
0060d000-0060e000 rw-p 0000d000 fe:01 10512978                           /usr/bin/idevicebackup2
00df0000-011af000 rw-p 00000000 00:00 0                                  [heap]
7f1838000000-7f1838021000 rw-p 00000000 00:00 0 
7f1838021000-7f183c000000 ---p 00000000 00:00 0 
7f183f30a000-7f183f320000 r-xp 00000000 fe:01 10489311                   /usr/lib/libgcc_s.so.1
7f183f320000-7f183f51f000 ---p 00016000 fe:01 10489311                   /usr/lib/libgcc_s.so.1
7f183f51f000-7f183f520000 r--p 00015000 fe:01 10489311                   /usr/lib/libgcc_s.so.1
7f183f520000-7f183f521000 rw-p 00016000 fe:01 10489311                   /usr/lib/libgcc_s.so.1
7f183f521000-7f183f522000 ---p 00000000 00:00 0 
7f183f522000-7f183fd22000 rw-p 00000000 00:00 0 
7f183fd22000-7f183fd24000 r-xp 00000000 fe:01 10489129                   /usr/lib/libdl-2.24.so
7f183fd24000-7f183ff24000 ---p 00002000 fe:01 10489129                   /usr/lib/libdl-2.24.so
7f183ff24000-7f183ff25000 r--p 00002000 fe:01 10489129                   /usr/lib/libdl-2.24.so
7f183ff25000-7f183ff26000 rw-p 00003000 fe:01 10489129                   /usr/lib/libdl-2.24.so
7f183ff26000-7f183ff2d000 r-xp 00000000 fe:01 10510554                   /usr/lib/libusbmuxd.so.4.0.0
7f183ff2d000-7f184012d000 ---p 00007000 fe:01 10510554                   /usr/lib/libusbmuxd.so.4.0.0
7f184012d000-7f184012e000 r--p 00007000 fe:01 10510554                   /usr/lib/libusbmuxd.so.4.0.0
7f184012e000-7f184012f000 rw-p 00008000 fe:01 10510554                   /usr/lib/libusbmuxd.so.4.0.0
7f184012f000-7f184037d000 r-xp 00000000 fe:01 10527446                   /usr/lib/libcrypto.so.1.0.0
7f184037d000-7f184057c000 ---p 0024e000 fe:01 10527446                   /usr/lib/libcrypto.so.1.0.0
7f184057c000-7f1840598000 r--p 0024d000 fe:01 10527446                   /usr/lib/libcrypto.so.1.0.0
7f1840598000-7f18405a4000 rw-p 00269000 fe:01 10527446                   /usr/lib/libcrypto.so.1.0.0
7f18405a4000-7f18405a7000 rw-p 00000000 00:00 0 
7f18405a7000-7f184060e000 r-xp 00000000 fe:01 10527445                   /usr/lib/libssl.so.1.0.0
7f184060e000-7f184080d000 ---p 00067000 fe:01 10527445                   /usr/lib/libssl.so.1.0.0
7f184080d000-7f1840811000 r--p 00066000 fe:01 10527445                   /usr/lib/libssl.so.1.0.0
7f1840811000-7f1840818000 rw-p 0006a000 fe:01 10527445                   /usr/lib/libssl.so.1.0.0
7f1840818000-7f18409ad000 r-xp 00000000 fe:01 10489065                   /usr/lib/libc-2.24.so
7f18409ad000-7f1840bac000 ---p 00195000 fe:01 10489065                   /usr/lib/libc-2.24.so
7f1840bac000-7f1840bb0000 r--p 00194000 fe:01 10489065                   /usr/lib/libc-2.24.so
7f1840bb0000-7f1840bb2000 rw-p 00198000 fe:01 10489065                   /usr/lib/libc-2.24.so
7f1840bb2000-7f1840bb6000 rw-p 00000000 00:00 0 
7f1840bb6000-7f1840bce000 r-xp 00000000 fe:01 10489008                   /usr/lib/libpthread-2.24.so
7f1840bce000-7f1840dcd000 ---p 00018000 fe:01 10489008                   /usr/lib/libpthread-2.24.so
7f1840dcd000-7f1840dce000 r--p 00017000 fe:01 10489008                   /usr/lib/libpthread-2.24.so
7f1840dce000-7f1840dcf000 rw-p 00018000 fe:01 10489008                   /usr/lib/libpthread-2.24.so
7f1840dcf000-7f1840dd3000 rw-p 00000000 00:00 0 
7f1840dd3000-7f1840de0000 r-xp 00000000 fe:01 10489050                   /usr/lib/libplist.so.3.0.0
7f1840de0000-7f1840fe0000 ---p 0000d000 fe:01 10489050                   /usr/lib/libplist.so.3.0.0
7f1840fe0000-7f1840fe1000 r--p 0000d000 fe:01 10489050                   /usr/lib/libplist.so.3.0.0
7f1840fe1000-7f1840fe2000 rw-p 0000e000 fe:01 10489050                   /usr/lib/libplist.so.3.0.0
7f1840fe2000-7f1841004000 r-xp 00000000 fe:01 10506478                   /usr/lib/libimobiledevice.so.6.0.0
7f1841004000-7f1841203000 ---p 00022000 fe:01 10506478                   /usr/lib/libimobiledevice.so.6.0.0
7f1841203000-7f1841204000 r--p 00021000 fe:01 10506478                   /usr/lib/libimobiledevice.so.6.0.0
7f1841204000-7f1841205000 rw-p 00022000 fe:01 10506478                   /usr/lib/libimobiledevice.so.6.0.0
7f1841205000-7f1841228000 r-xp 00000000 fe:01 10489057                   /usr/lib/ld-2.24.so
7f18413e9000-7f18413ef000 rw-p 00000000 00:00 0 
7f1841426000-7f1841427000 rw-p 00000000 00:00 0 
7f1841427000-7f1841428000 r--p 00022000 fe:01 10489057                   /usr/lib/ld-2.24.so
7f1841428000-7f1841429000 rw-p 00023000 fe:01 10489057                   /usr/lib/ld-2.24.so
7f1841429000-7f184142a000 rw-p 00000000 00:00 0 
7ffef4063000-7ffef4084000 rw-p 00000000 00:00 0                          [stack]
7ffef41e4000-7ffef41e6000 r--p 00000000 00:00 0                          [vvar]
7ffef41e6000-7ffef41e8000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

c0d3z3r0, 2016-11-22:
iPhone 6, iOS 10.0.2

cfergeau, 2016-11-26:
If this is on a linux machine, and if you can reproduce at will, installing debug packages for libimobiledevice and running "valgrind idevicebackup2 backup Stuff/iOSBackup/" should provide useful information for debugging this. Also, which version of libimobiledevice are you using?

c0d3z3r0, 2016-11-26:
I've tested both the latest release and master (commit 23069d1).
The backup seems to run fine with valgrind but crashes without as above o.O
[valgrind.log.txt](https://github.com/libimobiledevice/libimobiledevice/files/614138/valgrind.log.txt)

cfergeau, 2016-11-26:
Ah.. You could try to get a backtrace from gdb at least,
gdb --args idevicebackup2 backup Stuff/iOSBackup/
(gdb) run
<wait until the program crashes>
(gdb) thread apply all bt

I suspect debug symbols from libplist/libimobiledevice might be missing though.

c0d3z3r0, 2016-11-26:
Thread 2 (Thread 0x7ffff68f6700 (LWP 31465)):
#0  0x00007ffff6bdc0b3 in select () from /usr/lib/libc.so.6
#1  0x00007ffff72ca161 in ?? () from /usr/lib/libusbmuxd.so.4
#2  0x00007ffff72ca2fd in ?? () from /usr/lib/libusbmuxd.so.4
#3  0x00007ffff72c912d in usbmuxd_recv_timeout () from /usr/lib/libusbmuxd.so.4
#4  0x00007ffff7bbfdb7 in idevice_connection_receive_timeout () from /usr/lib/libimobiledevice.so.6
#5  0x00007ffff7bc0420 in service_receive_with_timeout () from /usr/lib/libimobiledevice.so.6
#6  0x00007ffff7bc0881 in ?? () from /usr/lib/libimobiledevice.so.6
#7  0x00007ffff7bc5b17 in ?? () from /usr/lib/libimobiledevice.so.6
#8  0x00007ffff6ea0454 in start_thread () from /usr/lib/libpthread.so.0
#9  0x00007ffff6be37df in clone () from /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff7fba700 (LWP 31464)):
#0  0x00007ffff6b2e04f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff6b2f47a in abort () from /usr/lib/libc.so.6
#2  0x00007ffff6b6bc50 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007ffff6b71fe6 in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007ffff6b727de in _int_free () from /usr/lib/libc.so.6
#5  0x00007ffff70b8452 in ?? () from /usr/lib/libplist.so.3
#6  0x00007ffff70be9e8 in ?? () from /usr/lib/libplist.so.3
#7  0x00007ffff70bea1f in ?? () from /usr/lib/libplist.so.3
#8  0x00000000004057d6 in mb2_handle_list_directory (mobilebackup2=0x611250, message=0x627de0, backup_dir=0x7fffffffea22 "/home/user/Stuff/iOSBackup/") at idevicebackup2.c:1104
#9  0x0000000000408483 in main (argc=3, argv=0x7fffffffe6d8) at idevicebackup2.c:2046

cfergeau, 2016-11-26:
libplist debug package will be needed for the backtrace to be useful.

c0d3z3r0, 2016-11-26:
Thread 2 (Thread 0x7ffff68e9700 (LWP 11425)):
#0  0x00007ffff6bcf0b3 in select () from /usr/lib/libc.so.6
#1  0x00007ffff72c0775 in socket_check_fd (fd=4, fdm=FDM_READ, timeout=500) at socket.c:323
#2  0x00007ffff72c09f6 in socket_receive_timeout (fd=4, data=0x7ffff68e8e84, length=4, flags=0, timeout=500) at socket.c:410
#3  0x00007ffff72bf7d1 in usbmuxd_recv_timeout (sfd=4, data=0x7ffff68e8e84 "", len=4, recv_bytes=0x7ffff68e8e48, timeout=500) at libusbmuxd.c:1130
#4  0x00007ffff7bb614c in internal_connection_receive_timeout (connection=0x6269c0, data=0x7ffff68e8e84 "", len=4, recv_bytes=0x7ffff68e8e48, timeout=500) at idevice.c:377
#5  0x00007ffff7bb62f0 in idevice_connection_receive_timeout (connection=0x6269c0, data=0x7ffff68e8e84 "", len=4, recv_bytes=0x7ffff68e8e48, timeout=500) at idevice.c:417
#6  0x00007ffff7bb7048 in service_receive_with_timeout (client=0x627de0, data=0x7ffff68e8e84 "", size=4, received=0x7ffff68e8e80, timeout=500) at service.c:161
#7  0x00007ffff7bb757c in internal_plist_receive_timeout (client=0x627720, plist=0x7ffff68e8f00, timeout=500) at property_list_service.c:187
#8  0x00007ffff7bb7981 in property_list_service_receive_plist_with_timeout (client=0x627720, plist=0x7ffff68e8f00, timeout=500) at property_list_service.c:255
#9  0x00007ffff7bbeabe in np_get_notification (client=0x613000, notification=0x7ffff68e8f40) at notification_proxy.c:258
#10 0x00007ffff7bbed84 in np_notifier (arg=0x629440) at notification_proxy.c:321
#11 0x00007ffff6e93454 in start_thread () from /usr/lib/libpthread.so.0
#12 0x00007ffff6bd67df in clone () from /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff7fba700 (LWP 11421)):
#0  0x00007ffff6b2104f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff6b2247a in abort () from /usr/lib/libc.so.6
#2  0x00007ffff6b5ec50 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007ffff6b64fe6 in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007ffff6b657de in _int_free () from /usr/lib/libc.so.6
#5  0x00007ffff70ab322 in hash_table_destroy (ht=0x6be4f0) at hashtable.c:51
#6  0x00007ffff70b3b6a in plist_free_data (data=0x627e60) at plist.c:191
#7  0x00007ffff70b3bc4 in plist_free_node (node=0x612ae0) at plist.c:205
#8  0x00007ffff70b3f3a in plist_free (plist=0x612ae0) at plist.c:312
#9  0x00000000004057d6 in mb2_handle_list_directory (mobilebackup2=0x610e00, message=0x627e00, backup_dir=0x7fffffffea34 "/home/c0d3/Stuff/iOSBackup/") at idevicebackup2.c:1104
#10 0x0000000000408483 in main (argc=3, argv=0x7fffffffe708) at idevicebackup2.c:2046

nikias, 2016-11-26:
Can you try this: in your libplist source go to src/plist.c and go to line 191. open a new line so you're after ```hash_table_destroy(data->hashtable);``` then add ```data->hashtable = NULL;```. Recompile and run again and see what happens.

c0d3z3r0, 2016-11-26:
Does not change anything:

Thread 2 (Thread 0x7ffff68e9700 (LWP 15385)):
#0  0x00007ffff6bcf0b3 in select () from /usr/lib/libc.so.6
#1  0x00007ffff72c0775 in socket_check_fd (fd=4, fdm=FDM_READ, timeout=500) at socket.c:323
#2  0x00007ffff72c09f6 in socket_receive_timeout (fd=4, data=0x7ffff68e8e84, length=4, flags=0, timeout=500) at socket.c:410
#3  0x00007ffff72bf7d1 in usbmuxd_recv_timeout (sfd=4, data=0x7ffff68e8e84 "", len=4, recv_bytes=0x7ffff68e8e48, timeout=500) at libusbmuxd.c:1130
#4  0x00007ffff7bb614c in internal_connection_receive_timeout (connection=0x613020, data=0x7ffff68e8e84 "", len=4, recv_bytes=0x7ffff68e8e48, timeout=500) at idevice.c:377
#5  0x00007ffff7bb62f0 in idevice_connection_receive_timeout (connection=0x613020, data=0x7ffff68e8e84 "", len=4, recv_bytes=0x7ffff68e8e48, timeout=500) at idevice.c:417
#6  0x00007ffff7bb7048 in service_receive_with_timeout (client=0x6147f0, data=0x7ffff68e8e84 "", size=4, received=0x7ffff68e8e80, timeout=500) at service.c:161
#7  0x00007ffff7bb757c in internal_plist_receive_timeout (client=0x627720, plist=0x7ffff68e8f00, timeout=500) at property_list_service.c:187
#8  0x00007ffff7bb7981 in property_list_service_receive_plist_with_timeout (client=0x627720, plist=0x7ffff68e8f00, timeout=500) at property_list_service.c:255
#9  0x00007ffff7bbeabe in np_get_notification (client=0x629650, notification=0x7ffff68e8f40) at notification_proxy.c:258
#10 0x00007ffff7bbed84 in np_notifier (arg=0x629440) at notification_proxy.c:321
#11 0x00007ffff6e93454 in start_thread () from /usr/lib/libpthread.so.0
#12 0x00007ffff6bd67df in clone () from /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff7fba700 (LWP 15381)):
#0  0x00007ffff6b2104f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff6b2247a in abort () from /usr/lib/libc.so.6
#2  0x00007ffff6b5ec50 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007ffff6b64fe6 in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007ffff6b657de in _int_free () from /usr/lib/libc.so.6
#5  0x00007ffff70ab322 in hash_table_destroy (ht=0x735c60) at hashtable.c:51
#6  0x00007ffff70b3b6a in plist_free_data (data=0x610540) at plist.c:191
#7  0x00007ffff70b3bcf in plist_free_node (node=0x612ae0) at plist.c:206
#8  0x00007ffff70b3f45 in plist_free (plist=0x612ae0) at plist.c:313
#9  0x00000000004057d6 in mb2_handle_list_directory (mobilebackup2=0x6112e0, message=0x627e00, backup_dir=0x7fffffffea36 "/home/c0d3/Stuff/iOSBackup/") at idevicebackup2.c:1104
#10 0x0000000000408483 in main (argc=3, argv=0x7fffffffe708) at idevicebackup2.c:2046

c0d3z3r0, 2016-11-26:
git bisect gave me https://github.com/libimobiledevice/libplist/commit/582c59bf7dcf37270c2fd7e99b4982ebc9bcbc74

nikias, 2016-11-26:
I reproduced the bug and am looking for a fix. Something is freeing the hash table before the plist_free call in mb2_handle_list_directory is trying to...

nikias, 2016-11-27:
Fixed with this commit in libplist: https://github.com/libimobiledevice/libplist/commit/950b6ddc3e503c4e1e3cb5d7813cdd9b41849cbc
Recompile libplist and retry. It should work now.

closed 2016-11-27 by nikias

c0d3z3r0, 2016-11-27:
Works now, thanks!