libimobiledevice/libimobiledevice #422: a probleme about idevicebackup2(the -list CMD)

RonnieLLL, 2016-11-23:
hi! I want get last last completed backup in CSV format, but there has a error!

CMD: -d -u 8050e9e4e0366abfec506be96dece1ef4afef887 list C:\testbackup

Error reading backup database (MBErrorDomain/205). Underlying error: No such file or directory at path "8050e9e4e0366abfec506be96dece1ef4afef887/Manifest.mbdb" (MBErrorDomain/4).

But In the backup folder, there is no Manifest.mbdb only Manifest.db 

Test environment: iPad Air  iOS10.1.1  windows 7!

Is there any way to solve it?

aymenim, 2016-11-24:
My guess is Apple has not updated the Api to handle iOS 10 backups, because starting on iOS 10 the mbdb file structure was replaced by a sqlite 3 database. the solution is to see if apple updates it or look if there is a replacement Api.

nikias, 2016-12-15:
Yes. Apple messed it up (again), sorry.

closed 2016-12-15 by nikias

BlueManCZ, 2020-01-04:
Is there any solution for this? I have the same problem when trying to list backup of iPhone 5.

mexmer, 2020-01-06:
no, there is no solution, apple didn't update their iOS tools to work properly with newer version of iOS backup. since iOS 10, when they started using manifest.db it stopped working. even their official tools (shipped with itunes) don't work. only supported functionality is backup and restore.

BlueManCZ, 2020-01-06:
> only supported functionality is backup and restore.

Oh, well. At least something. Thank you for answer @mexmer