libimobiledevice/libimobiledevice #429: idevicebackup 2 restore from old device backup

iOSShohrab, 2016-12-06:
I am trying to restore my one device from my old device back up. 
first i back up my old device using idevicebackup2 backup PATH
then i try to restore it but having an error saying, info.plist has issue. i understand its because of UUID and IMEI isn't matching.
can you please guide me how to restore my old device with my old device backup
thanks

mexmer, 2016-12-07:
IMEI and UUID should be no issue, ios allows cross device restore, only requirement is, that TARGET device iOS version is same or newer as iOS version of device in backup.

eg. you cannot restore into iOS 8.0, backup you made from device with iOS 8.1.

iOSShohrab, 2016-12-07:
thanks for your reply. i check both iOS version which is 10.1.1 here is the steps what i did >>

1. first i take a back up using` idevicebackup2 backup PATH` ans it creates a back up folder using its UDID.
2. Now i connect with my new device and tried command idevicebackup2 restore FOLDER_PATH. it saying an ERROR occured. NO info.plist found for UDID <device_udid>

what is the command to restore from old device back up.

nikias, 2016-12-07:
@iOSShohrab you need to pass the UDID of the source (old) device:
```
idevicebackup2 --source UDID restore --system --settings --reboot PATH
```

iOSShohrab, 2016-12-07:
thats great. its work now. really save my life man :)
is there any way to skip activation part?

mexmer, 2016-12-08:
no, you cannot skip activation part.
and no, application restore is not supported for iOS9 and newer, but should be possible see #343

iOSShohrab, 2016-12-08:
thanks. appreciate for your response