libimobiledevice/libimobiledevice #432: ideviceprovision can't parse development provisioning profiles

oclbdk, 2016-12-08:
# Impact

This bug prevents people from seeing the complete list of provisioning profiles installed on a device. Without being able to see the provisioning profiles UUID's, it's very difficult to remove them. The device will slowly accumulate more and more provisioning profiles and slow down app startup if the developer is debugging many different apps

# Repro steps

1. Pair a device
2. Install a development provisioning profile (e.g. https://github.com/magbetworksforfood/Snatris/blob/8cb067421fe61b8a4be8a0cc4767433c0ac7cfc0/testprofile.mobileprovision -- you'd need to generate your own with your device's UDID, otherwise it'll fail to install in the first place)
3. `ideviceprovision list` has the following output where I expect to see the provisioning profile UUID and name:

```
profile_get_embedded_plist: unexpected profile data (1)
(unknown id) - (no name)
```

# Fix

The code is failing on this line: https://github.com/libimobiledevice/libimobiledevice/blob/9df4e695c55ddd319c5acaf7fa21f960afaa1e90/tools/ideviceprovision.c#L139

There seems to be some assumption with the size of the plist data that only holds for distribution provisioning profiles?

nikias, 2016-12-15:
It works for me. Can you add
```
printf("blen=%ld slen=%d\n", blen, slen);
```
before line 139 and run it with your device and the given profile installed?

mexmer, 2016-12-16:
i suspect you should use 
```c
printf("blen=%lld slen=%lld\n", blen, slen);
```
just in case, there is something "SO wrong" with numbers. you using 64bit so %ld is not enough.

oclbdk, 2016-12-16:
I added this line:

```
printf("blen=%llu slen=%hu\n", blen, slen);
```

I got this output for the unparseable provisioning profile:

```
blen=1908946 slen=7456
profile_get_embedded_plist: unexpected profile data (1)
(unknown id) - (no name)
```

nikias, 2016-12-17:
1908946 is 0x1D20D2 in hex, and 7456 is 0x1D20. It seems as if it's reading too many bytes for the value which is supposed to be 16 bit.

nikias, 2016-12-17:
The large value comes from libplist's plist_get_data_val(). Somehow it seems to get too much data?!

oclbdk, 2016-12-17:
Ha yeah, we have really big provisioning profiles. This error makes sense,
thanks for investigating. Would it be fixable by using a bigger integer?

nikias, 2016-12-17:
Can you show me the first 16 bytes of the provisioning profile? e.g.:
```
hexdump profile.mobileprovision |head -1
```
Then I can add support for it.

oclbdk, 2016-12-17:
Yeah happy to do it on Monday (I left my computer at work)

As a sidenote, thanks for building such a useful tool and being so
responsive on these issues!

oclbdk, 2016-12-19:
0000000 30 83 1d 20 cd 06 09 2a 86 48 86 f7 0d 01 07 02

nikias, 2016-12-21:
@oclbdk can you try this patch: https://gist.github.com/nikias/662655ea7ffccbd891ec79e9004ac7f1

oclbdk, 2016-12-21:
Thanks! Unfortunately, the same error occurs

oclbdk, 2016-12-21:
Thanks! Unfortunately, the same error occurs

If it helps, I noticed that it always go through this conditional path

```
+	if (bsize & 0x80) {
+		slen += 2 + bsize & 0xF;
+	}
```

I tried hardcoding it to always go through the else

```
+	} else {
+		slen += 3;
+	}
```

but that doesn't parse the provisioning profile correctly either

nikias, 2016-12-21:
Ok. I sent you an email. Please check.

oclbdk, 2016-12-22:
I looked into this and found that the bitshift is on the wrong end for `case 3` -- `res = ntohl(ds >> 8);` should be `res = ntohl(ds) >> 8;`

nikias, 2016-12-22:
Fixed with commit a80ba9363b4dc0ea1bcdad152cf2cdc974636ad6.

closed 2016-12-22 by nikias