libimobiledevice/libimobiledevice #433: ifuse /media/iPhone --root:  GnuTLS error: Error in the pull function. Failed to connect to lockdownd service on the device. Try again. If it still fails try rebooting your device.

starrychloe, 2016-12-08:
I tried to follow the guide here
http://www.dedoimedo.com/computers/linux-iphone-6.html
referenced here
http://superuser.com/questions/1154061/how-do-i-copy-photos-from-my-iphone-on-ubuntu-16-04
and it failed with this command
```
chloe@chloe-latitude-d430:/$ sudo ifuse /media/chloe/iPhone --root
GnuTLS error: Error in the pull function.
Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.
```
I tried to reboot the iPhone but it didn't work. The pair command worked.
```
chloe@chloe-latitude-d430:/$ idevicepair pair
SUCCESS: Paired with device xxxxxxxxxx4c5c0f1467f0fcxxxxxxxxxxxxxx
```
I already have the latest `ifuse`
```
chloe@chloe-latitude-d430:/$ apt show ifuse
Package: ifuse
Version: 1.1.2-0.1build3
Priority: optional
Section: universe/utils
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Julien Lavergne <julien.lavergne@gmail.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 44.0 kB
Depends: fuse, libc6 (>= 2.4), libfuse2 (>= 2.8), libimobiledevice6 (>= 1.1.0), libplist3 (>= 1.11)
Homepage: http://libimobiledevice.org
Task: ubuntu-mate-desktop
Supported: 3y
Download-Size: 13.8 kB
APT-Manual-Installed: yes
APT-Sources: http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
Description: FUSE module for iPhone and iPod Touch devices
 iFuse is a FUSE filesystem driver which uses libiphone to connect to iPhone
 and iPod Touch devices without needing to "jailbreak" them. iFuse uses the
 native Apple AFC protocol over a normal USB cable in order to access the
 device's media files.
 .
 Although iFuse is now in a working state it is still under heavy
 development and should be considered experimental.

chloe@chloe-latitude-d430:/$ sudo apt install ifuse
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ifuse is already the newest version (1.1.2-0.1build3).
```

Ubuntu 16.04, iPhone 5, iOS 10.1.1

dullroar, 2016-12-10:
Getting the same. Linux Mint 18, 4.4.0-21 kernel, same version of ifuse as reported above, iPhone 6s, iOS 10.1.1.

borisovg, 2016-12-12:
This is probably the same as  #413?

dullroar, 2016-12-12:
It would seem so, @borisovg. I note there is supposed to be a bug fix in the code base for that, but the latest release is 1.2.0 from Jan. 28, 2015 (which is what I am running), so I presume I will have to clone and build the lib myself to get the patch. Will queue it up for when I get some free time.

dullroar, 2016-12-13:
Aaaaannnd...cloned, rebuilt (took some dependency installs, but I got there) and installed, and same error.

borisovg, 2016-12-13:
I can confirm that building `libimobiledevice` and `usbmuxd` make it work for me (as tested with `ideviceinfo`) but I had to remove the relevant Debian packages first.

Are there any plans for a release soon - I doubt we will get a new package in Debian without it. And with the freeze for the upcoming stable coming up very soon it would be great to get this fix into it!

dullroar, 2016-12-13:
@nikias I did do an unpair and re-pair after I rebuilt the lib. @borisovg I wanted to try removing the existing packages first, but under Mint everything and its dog has this lib as a dependency and it was warning it was going to uninstall Cinnamon(!) and a whole bunch of other packages (see screenshot).
[image]
I am sure I can more "surgically" remove the libs with apt, so will try that tonight, hopefully.

borisovg, 2016-12-13:
@dullroar I feel your pain - I also had to drop `gvfs-backends` which I'm not happy about. But the alternative is not being able to get at my phone's photos and videos. :(

The real solution is for a release to be published and packaged by Debian, which should in turn filter down to Mint and Ubuntu.

haydenbjyoung, 2016-12-25:
Hello everybody, I have just made a gist to install libimobile and openssl which (for me) fixed the issue with the lockdownd error.

samrocketman, 2016-12-27:
@dullroar did you build based on https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

?

dullroar, 2016-12-27:
@samrocketman I only ended up rebuilding `libimobiledevice` plus its pre-reqs (I've slept since then and don't recall everything I had to build to get it to work).

I have moved on, given that the real problem I was trying to solve was getting some images and videos off the iPhone that I ended up unfortunately having to delete, since not even sending them to iCloud or sharing via email or messaging from the phone itself worked for them, and neither did syncing with iTunes on Windows. In all cases the images/videos ended up being not accessible, even to Apple products/iOS itself, even though I could see their thumbnails on the phone. In most cases I had already gotten the images off the phone before the problem showed up by using Dropbox, but in a few cases I just lost them. Sigh,

eura665, 2016-12-31:
This ppa work with iOS 10.1 and lower https://launchpad.net/~martin-salbaba/+archive/ubuntu/ppa+libimobiledevice but not with iOS 10.2 ;-( then you are back to the errors. Try it on LinuxMint 18.1 Hope someone can fix it.