libimobiledevice/libimobiledevice #436: Update README regarding OpenSSL requirement

borisovg, 2016-12-12:
README currently says:

```
Requirements
============

Development Packages of:
libgnutls or openssl
```

But on running `./autogen.sh` I get:

```
checking for openssl... no
configure: error: OpenSSL support explicitly requested but OpenSSL could not be found
```

So I guess README needs to be updated?

samrocketman, 2016-12-12:
For using GNU TLS it needs to be built with:

```
./autogen.sh --disable-ssl
```
The option changed since I've built it so it may be `--disable-openssl`.

See docs for `autogen.sh` to be sure.

```
./autogen.sh 2>&1 | grep -- ssl
```

borisovg, 2016-12-13:
Can this information be added to the README's build instructions?

gitsop01, 2016-12-14:
It should also have a comments on what version of Openssl libary it has been tested on, as openssl version is now up to 1.10. But most Linux distributions are not there yet, Fedora is going to incorporate
Openssl 1.10 in the near future.

nikias, 2017-04-28:
Updated README with commit 0cf6bb6.

closed 2017-04-28 by nikias

yaochx, 2019-07-03:
./autogen.sh --disable-ssl
sudo apt install libgnutls28-dev

HD421, 2020-05-25:
Worked for me (Ubuntu 16.04):
sudo apt-get install libgcrypt-dev
sudo apt-get install gnutls-bin
./autogen.sh --disable-ssl