libimobiledevice/libimobiledevice #438: URGENT: idevicebackup2 error -6 on restore and unback

ryanexec, 2016-12-14:
This is an encrypted backup.
Enter backup password: *********
Started "com.apple.mobilebackup2" service on port 49742.
Could not perform backup protocol version exchange, error code -6

-Backup from 6S on 10.1.1
-Restore to 6S on 10.1.1
-Win10x64
-Latest iTunes
-iMobileDevice 1.2.1.81 for Win x64 by Quamotion (libimobiledevice ported to Win)
-Tried with iTunes open & closed (requires iTunes until iFuse port to Win is ready)
-Tried disabling FW & AV
-Any documentation on error -6?

Currently lost all data, unable to complete the restore in iTunes, and I cannot find sufficient iTunes logging to help target and troubleshoot why the iTunes restore is failing. iTunes restore processes about halfway before failing, but iDeviceBackup2 fails immediately.

nikias, 2016-12-15:
error -6 means MOBILEBACKUP2_E_NO_COMMON_VERSION, see: https://github.com/libimobiledevice/libimobiledevice/blob/master/include/libimobiledevice/mobilebackup2.h#L44

The error code is set here:
https://github.com/libimobiledevice/libimobiledevice/blob/master/src/mobilebackup2.c#L304

This error condition comes from the device. Have you tried running idevicesyslog and look for "BackupAgent2" messages?

ryanexec, 2016-12-15:
Nikias, thank you. I had eventually found those, but no insight into how to troubleshoot them. I will try to follow up on your idevicesyslog suggestion and try to figure that part out, but how would that apply to not being able to unback the backup due to error -6? How would I go about troubleshooting the unback process??

nikias, 2016-12-15:
The device isn't happy for some reason. Check idevicesyslog to see what's going on.

ryanexec, 2016-12-15:
But unback should be device independent, no? Unless you mean my computer, which I will check the event logs for any signs of why it fails after trying to open the port. This seems to be computer-side, but I am trying to understand the tools better, and any logging they can provide.

ryanexec, 2016-12-15:
On another note, I'm currently using iDeviceSysLog to monitor another iTunes restore attempt, to try and identify where the iTunes restore process is failing, and on what files.

nikias, 2016-12-15:
Please note that unback (and also list) is broken on iOS 10+ thanks to Apple.

ryanexec, 2016-12-15:
okay, so I cannot extract my 10.1.1 backup. sadface. i should still be able to restore my 10.1.1 backup, correct? but both processes fail immediately with the same error.

I'm assuming from the below, I have a port/protocol being blocked, or I'm missing a dependency??

Dec 14 18:33:32 Labs-iPhone pipelined[264] <Info>: os_transaction.h(40): [OSTransaction] Ending transaction for keybag unlocked
Dec 14 18:33:32 Labs-iPhone BackupAgent2(libsystem_trace.dylib)[416] <Notice>: subsystem: com.apple.mobilebackup, category: Default, enable_level: 2, persist_level: 1, default_ttl: 14, info_ttl: 14, debug_ttl: 0, generate_symptoms: 0, enable_oversize: 0, privacy_setting: 0, enable_private_data: 0
Dec 14 18:33:32 Labs-iPhone BackupAgent2(libsystem_trace.dylib)[416] <Notice>: subsystem: com.apple.mobilebackup, category: Trace, enable_level: 2, persist_level: 1, default_ttl: 14, info_ttl: 14, debug_ttl: 0, generate_symptoms: 0, enable_oversize: 0, privacy_setting: 0, enable_private_data: 0
Dec 14 18:33:32 Labs-iPhone coreduetd[90] <Info>: kMobileKeyBagLockStatusNotificationID state changed to 0
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Notice>: Device UDID: <private>
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Checking in with lockdown

Dec 14 18:33:32 Labs-iPhone BackupAgent2(liblockdown.dylib)[416] <Info>: Checked in with lockdown on path 'xpc' socket 4 without SSL
Dec 14 18:33:32 Labs-iPhone BackupAgent2(liblockdown.dylib)[416] <Info>: XPC listener can

Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Creating device connection
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Waiting for connection from computer
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Running connection
Dec 14 18:33:32 Labs-iPhone routined(libcoreroutine.dylib)[247] <Info>: --- handle event <private> ---
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Received message: <private>
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Error>: No common supported protocol version
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Sending response message: 1 No common supported protocol version
Dec 14 18:33:32 Labs-iPhone softwareupdateservicesd(SoftwareUpdateServices)[184] <Info>: Device lock status changed: 3
Dec 14 18:33:32 Labs-iPhone softwareupdateservicesd(SoftwareUpdateServices)[184] <Info>: Device unlocked
Dec 14 18:33:32 Labs-iPhone BackupAgent2[416] <Info>: Computer disconnected
Dec 14 18:33:32 Labs-iPhone IDSKeychainSyncingProxy[101] <Notice>: keybagStateChange! was locked: 0

ryanexec, 2016-12-15:
Here's the death of the iTunes restore process:

Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloading 31 files of size 926 B
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (18.8 B at 2.82 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (11.7 B at 188 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (39 B at 18.8 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (12.4 B at 196 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (8.05 B at 141 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (10.2 B at 173 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (84.8 B at 41.1 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (10.2 B at 138 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (18.9 B at 15 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (11.7 B at 182 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (6.58 B at 120 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (10.3 B at 178 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (22.5 B at 12.5 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (41 B at 22 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (23.4 B at 15.2 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (20.6 B at 12.4 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (22.5 B at 14 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (40 B at 20.9 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (9.62 B at 158 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (91.7 B at 57.3 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (16 B at 160 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (6.66 B at 119 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (41 B at 20.5 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (208 B at 91.3 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (16.1 B at 161 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (13.4 B at 206 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (62.2 B at 102 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (10.2 B at 176 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (17.1 B at 14.5 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (10.1 B at 116 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Downloaded: <private> (10.9 B at 175 kB/s)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file: <private> (<private>)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file ownership: 501:501
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file permissions: 0644
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring last modified time (<private>) for <private>
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring protection class: 3
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Renaming temporary file
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file: <private> (<private>)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file ownership: 501:501
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file permissions: 0644
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring last modified time (<private>) for <private>
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring protection class: 3
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Renaming temporary file
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file: <private> (<private>)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file ownership: 501:501
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file permissions: 0644
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring last modified time (<private>) for <private>
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring protection class: 3
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Renaming temporary file
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file: <private> (<private>)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file ownership: 501:501
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file permissions: 0644
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring last modified time (<private>) for <private>
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring protection class: 3
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Renaming temporary file
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file: <private> (<private>)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file ownership: 501:501
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file permissions: 0644
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring last modified time (<private>) for <private>
[... cut ...]
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file permissions: 0644
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring last modified time (<private>) for <private>
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring protection class: 3
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Renaming temporary file
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file: <private> (<private>)
Dec 14 18:45:35 sToic BackupAgent2[422] <Info>: Restoring regular file ownership: 501:501
D[disconnected]

ryanexec, 2016-12-15:
All mentions of ReportCrash during an iTunes restore, full log zipped & attached. Real issue remains why I can't get iDeviceBackup2 to launch successfully; what would the protocol be that it cannot find??

Dec 14 19:19:24 sToic BackupAgent2[213] <Notice>: Finished copying backup in 216.145 s at 64.5 kB/s
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Updating backup info
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Writing <private>
Dec 14 19:19:24 sToic ReportCrash(CrashReporterSupport)[220] <Notice>: cr_update: <private>
Dec 14 19:19:24 sToic ReportCrash(CrashReporterSupport)[220] <Notice>: cr_update: <private>
Dec 14 19:19:24 sToic BackupAgent2[213] <Notice>: Finished preparing for restore in 269.908 s
Dec 14 19:19:24 sToic BackupAgent2[213] <Notice>: Starting to restore 51306 files of size 13.4 MB

Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Making directory at <private> (0700)
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Restoring directory ownership: 501:501
Dec 14 19:19:24 sToic ReportCrash[220] <Notice>: Formulating report for corpse[213] <private>
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Restoring directory: <private> (<private>)
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Making directory at <private> (0755)
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Restoring directory ownership: 501:501
Dec 14 19:19:24 sToic ReportCrash(CrashReporterSupport)[220] <Notice>: report not saved because it is non-actionable
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Creating directory: <private>
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Restoring directory: <private> (<private>)
Dec 14 19:19:24 sToic BackupAgent2[213] <Info>: Directory already exists

Dec 14 19:20:46 sToic BackupAgent2[213] <Info>: Renaming temporary file
Dec 14 19:20:46 sToic ReportCrash(CrashReporterSupport)[221] <Notice>: cr_update: <private>
Dec 14 19:20:46 sToic ReportCrash(CrashReporterSupport)[221] <Notice>: cr_update: <private>
Dec 14 19:20:46 sToic BackupAgent2[213] <Info>: Restoring regular file: <private> (<private>)

Dec 14 19:20:46 sToic BackupAgent2[213] <Info>: Renaming temporary file
Dec 14 19:20:46 sToic ReportCrash[221] <Notice>: Formulating report for corpse[207] <private>
Dec 14 19:20:46 sToic ReportCrash(CrashReporterSupport)[221] <Notice>: report not saved because it is non-actionable
Dec 14 19:20:46 sToic BackupAgent2[213] <Info>: Downloading 31 files of size 352 B

[iDeviceSysLog_Restore.zip](https://github.com/libimobiledevice/libimobiledevice/files/653706/iDeviceSysLog_Restore.zip)

ryanexec, 2016-12-15:
So, booted up a fresh Ubuntu 16.10 usb just remove Windows from the equation, but I'm getting the lockdownd error- 5 when trying to connect with ideviceinfo. Trying to find the resolution for that, but not seeing it.

ryanexec, 2016-12-15:
Thank you again for your responses earlier. Update: I worked through all the issues, and finally got iDeviceBackup2 to actually start a restore. It's now dying at 14% with the below message and iDeviceSysLog of restore attempt. If I can understand what file(s) it's tripping over, I'm hoping to try removing them and seeing if that helps.

Sending '6afb310aff20d86b0a3329303ee1e6294dd64f7c/a2/a2d0269fd028df5ae1eaac7855bbe81039416796' (288 Bytes)
 [========                                          ]  14% Finished
ErrorCode 205: Multiple errors downloading files (MBErrorDomain/205)
The device should reboot now.
Restore Failed (Error Code 205).

[iDeviceSysLog_Restore_Ubuntu.txt.zip](https://github.com/libimobiledevice/libimobiledevice/files/656080/iDeviceSysLog_Restore_Ubuntu.txt.zip)