libimobiledevice/libimobiledevice #440: [help] compile error in macOS 10.12.1 using latest libimobiledevice code

krisdyZJU, 2016-12-19:
I search in the history issues list, not found the similar error log like mine.

error code below when I try to execute 'make' in libimobiledevice folder:
`Making all in common
make[2]: Nothing to be done for `all'.
Making all in src
  CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_plist_access_path", referenced from:
      _instproxy_client_get_path_for_bundle_identifier in installation_proxy.o
  "_plist_array_append_item", referenced from:
      _device_link_service_version_exchange in device_link_service.o
      _device_link_service_disconnect in device_link_service.o
      _device_link_service_send_ping in device_link_service.o
      _device_link_service_send_process_message in device_link_service.o
      _file_relay_request_sources_timeout in file_relay.o
      _instproxy_append_current_list_to_result_cb in installation_proxy.o
      _instproxy_lookup in installation_proxy.o
      ...
  "_plist_array_get_item", referenced from:
      _device_link_service_version_exchange in device_link_service.o
      _device_link_service_get_message in device_link_service.o
      _device_link_service_receive_process_message in device_link_service.o
      _lockdownd_get_sync_data_classes in lockdown.o
      _instproxy_append_current_list_to_result_cb in installation_proxy.o
      _mobilesync_start in mobilesync.o
      _mobilesync_finish in mobilesync.o
      ...
  "_plist_array_get_size", referenced from:
      _device_link_service_version_exchange in device_link_service.o
      _device_link_service_get_message in device_link_service.o
      _device_link_service_receive_process_message in device_link_service.o
      _instproxy_status_get_current_list in installation_proxy.o
      _mobilebackup_send_restore_complete in mobilebackup.o
      _plist_dict_print_to_stream in libinternalcommon.a(utils.o)
      _plist_array_print_to_stream in libinternalcommon.a(utils.o)
      ...
  "_plist_copy", referenced from:
      _device_link_service_send_process_message in device_link_service.o
      _device_link_service_receive_process_message in device_link_service.o
      _lockdownd_get_value in lockdown.o
      _lockdownd_do_pair in lockdown.o
      _lockdownd_activate in lockdown.o
      _lockdownd_build_start_service_request in lockdown.o
      _instproxy_browse_with_callback in installation_proxy.o
      ...
  "_plist_dict_get_item", referenced from:
      _lockdown_check_result in lockdown.o
      _lockdownd_query_type in lockdown.o
      _lockdownd_get_value in lockdown.o
      _lockdownd_start_session in lockdown.o
      _lockdownd_do_pair in lockdown.o
      _lockdownd_do_start_service in lockdown.o
      _lockdownd_build_start_service_request in lockdown.o
      ...
  "_plist_dict_new_iter", referenced from:
      _plist_dict_print_to_stream in libinternalcommon.a(utils.o)
  "_plist_dict_next_item", referenced from:
      _plist_dict_print_to_stream in libinternalcommon.a(utils.o)
  "_plist_dict_remove_item", referenced from:
      _lockdownd_do_pair in lockdown.o
  "_plist_dict_set_item", referenced from:
      _lockdownd_stop_session in lockdown.o
      _plist_dict_add_label in lockdown.o
      _lockdownd_query_type in lockdown.o
      _lockdownd_get_value in lockdown.o
      _lockdownd_set_value in lockdown.o
      _lockdownd_remove_value in lockdown.o
      _lockdownd_pair in lockdown.o
      ...
[... 114 lines cut ...]
  "_plist_to_bin", referenced from:
      _internal_plist_send in property_list_service.o
      _webinspector_send in webinspector.o
      _userpref_save_pair_record in libinternalcommon.a(userpref.o)
      _plist_write_to_filename in libinternalcommon.a(utils.o)
  "_plist_to_xml", referenced from:
      _internal_plist_send in property_list_service.o
      _plist_write_to_filename in libinternalcommon.a(utils.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2`

the environment situation of my mac:
macOS Sierra 10.12.1
XCode version: 8.2

bash profile related:
`export libplist_CFLAGS='-I/usr/local/opt/libplist/include'
export libplist_LIBS='-L/usr/local/opt/libplist/lib'
export libplistmm_CFLAGS='-I/usr/local/opt/libplist/include'
export libplistmm_LIBS='-L/usr/local/opt/libplist/lib'
export CFLAGS='-I/usr/local/opt/openssl/include'
export LDFLAGS='-L/usr/local/opt/openssl/lib'`

krisdyZJU, 2016-12-19:
my openssl version is : OpenSSL 1.0.2h

krisdyZJU, 2016-12-19:
@nikias @FunkyM 
really sorry to bother you, but can you pls help to take a look, thanks

mexmer, 2016-12-19:
your build is missing libplist. did you run configure, and did it find libplist?

krisdyZJU, 2016-12-20:
@mexmer yes, I run ./autogen.sh before I run make, and it can find libplist and libplistmm both: 

> .....
checking for libplist... yes
checking for libplistmm... yes
...........
Configuration for libimobiledevice 1.2.1:
-------------------------------------------

  Install prefix: .........: /usr/local
  Debug code ..............: no
  Python bindings .........: no
  SSL support backend .....: OpenSSL

  Now type 'make' to build libimobiledevice 1.2.1,
  and then 'make install' for installation.

krisdyZJU, 2016-12-20:
Fixed now.
Previously, I use brew to install libplist. I just uninstall libplist and compile it from scratch using source code, and it works now.
So I just close it, thanks all.

closed 2016-12-20 by krisdyZJU