libimobiledevice/libimobiledevice #441: Win 32 - idevicebackup2 issue

iOSShohrab, 2016-12-19:
Hi,
I tried to build using VS 2015 in my Win machine. After complete build, when I tried to create backup of my machine using ibackup2 it showing this error (screen shot)
```

C:\Users\Admin\Downloads\libimobiledevice-msvc-master\Win32\Debug>idevicebackup2 backup C:\Users\Admin\Desktop\Compile_Try
Backup directory is "C:\Users\Admin\Desktop\Compile_Try"
Started "com.apple.mobilebackup2" service on port 49210.
Could not perform backup protocol version exchange, error code -6
```

i build, libplist,  libusbmuxd, libmobiledevice.

[image]

mexmer, 2016-12-19:
pair your device with idevicepair first

iOSShohrab, 2016-12-19:
i try with Pair, showing` ERROR: Could not connect to lockdownd, error code -21`
```
C:\Users\Admin\Downloads\libimobiledevice-msvc-master\Win32\Debug>idevicepair pair
SUCCESS: Paired with device 393f05095cdc8849ac222801a3845876c0ea94e6

C:\Users\Admin\Downloads\libimobiledevice-msvc-master\Win32\Debug>idevicebackup2.exe backup C:\Users\Admin\Desktop\Compile_Try
Backup directory is "C:\Users\Admin\Desktop\Compile_Try"
ERROR: Could not connect to lockdownd, error code -21
```

mexmer, 2016-12-19:
Could not connect to lockdownd, error code -21 means either device denied pairing (but since you paired, should be ok), or invalid pairing information.

i suspect, there might be problem reading pairing information.
try running it as administrator.

other possibility (if you compiled it with gnutls and not openssl), there has been bug in gnutls generated certificates, but should be fixed, so make sure, you have current sourcecode.

iOSShohrab, 2016-12-19:
i am trying to build code using https://github.com/libimobiledevice-win32/libimobiledevice
i tried to build using visual studio 2015. 
what i did, just open visual studio 2015 > restore nugets > then build.
no luck yet.

iOSShohrab, 2016-12-19:
for more info >>
i actually stuck in here. getting err from code

```
/* send Hello message */
		double local_versions[2] = {2.0, 2.1};
		double remote_version = 0.0;
		err = mobilebackup2_version_exchange(mobilebackup2, local_versions, 2, &remote_version);
		if (err != MOBILEBACKUP2_E_SUCCESS) {
			printf("Could not perform backup protocol version exchange, error code %d\n", err);
			cmd = CMD_LEAVE;
			goto checkpoint;
		}
```

mexmer, 2016-12-20:
never used that source, so don't know, but -21 usually means pairing record error, you can find multiple issues here with that error

if you compile against gnutls, make sure this patch is present in code
https://github.com/libimobiledevice/libimobiledevice/commit/23069d10341ce637fdad7321d447c53752dba48c

seast123, 2017-04-21:
Could not perform backup protocol version exchange, error code -6

seast123, 2017-04-21:
"Could not perform backup protocol version exchange, error code ",how to solve?help~~

nikias, 2017-05-07:
It must have to do with how you build the libraries and tools. I compiled everything with mingw on windows and had no issues when testing with a 10.3.1 device.

qmfrederik, 2017-08-18:
I maintain that for of libimobiledevice. The issue was that libplist used an incorrect endianness to serialize float values in binary property lists. https://github.com/libimobiledevice-win32/libplist/commit/40aaaaae633f52e7b858c743162dc26720662211 fixes it.
libimobiledevice-win32/libimobiledevice tree has also been updated to use the latest version of libplist, so you should be able to just do git update, nuget restore and rebuild.