libimobiledevice/libimobiledevice #442: what is the difference between two git repository url behind?

krisdyZJU, 2016-12-20:
A small question... but i really do not know how to understand:
we can see there are two git repository url in the readme.txt of libimobiledevice, showed below:

> Code:
	git clone http://git.libimobiledevice.org/libimobiledevice.git

Code (Mirror):
	git clone https://github.com/libimobiledevice/libimobiledevice.git

so what is the difference between them? I see the file count is different, are they the same?
will they be synchronized automatically or manually?

nikias, 2016-12-21:
One of them is a mirror as the README says. the libimobiledevice.org repository might contain some experimental branches from time to time. But master is always in sync.

closed 2016-12-21 by nikias

krisdyZJU, 2016-12-21:
got it, thanks.