libimobiledevice/libimobiledevice #443: Make Error

daduuuu, 2016-12-30:
Can someone help me with this error

./autogen.sh
glibtoolize: putting auxiliary files in '.'.
glibtoolize: linking file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: linking file 'm4/libtool.m4'
glibtoolize: linking file 'm4/ltoptions.m4'
glibtoolize: linking file 'm4/ltsugar.m4'
glibtoolize: linking file 'm4/ltversion.m4'
glibtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:31: installing './compile'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
configure: WARNING: Libtool does not cope well with whitespace in `pwd`
checking build system type... x86_64-apple-darwin16.4.0
checking host system type... x86_64-apple-darwin16.4.0
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
checking how to convert x86_64-apple-darwin16.4.0 file names to x86_64-apple-darwin16.4.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin16.4.0 file names to toolchain format... func_convert_file_noop
checking for /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
[... 141 lines cut ...]
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_num_locks", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_set_id_callback", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_set_locking_callback", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_EVP_cleanup", referenced from:
      _internal_idevice_deinit in idevice.o
  "_SSL_library_init", referenced from:
      _internal_idevice_init in idevice.o
  "_X509_set_notAfter", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_X509_set_notBefore", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_sk_free", referenced from:
      _internal_idevice_deinit in idevice.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

daduuuu, 2017-01-02:
nvm I solved it

closed 2017-01-02 by daduuuu

conath, 2017-02-02:
@daduuuu How did you solve it?