libimobiledevice/libimobiledevice #444: Trap abort 6

BlackGeekTutorial, 2016-12-30:
I compiled the latest source of libxml, libplist, libusbmuxd, usbmuxd and libimobiledevice from this official source.
As soon as I run every idevice* cmd, I get this error:

MBP-di-Andrea:libimobiledevice benti$ ideviceinfo --help
Usage: ideviceinfo [OPTIONS]
Show information about a connected device.

  -d, --debug		enable communication debugging
  -s, --simple		use a simple connection to avoid auto-pairing with the device
  -u, --udid UDID	target specific device by its 40-digit device UDID
  -q, --domain NAME	set domain of query to NAME. Default: None
  -k, --key NAME	only query key specified by NAME. Default: All keys.
  -x, --xml		output information as xml plist instead of key/value pairs
  -h, --help		prints usage information

Homepage: <http://libimobiledevice.org>
**ideviceinfo(65939,0x7fffae4713c0) malloc: *** error for object 0x7f98934071f0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6**

iFred09, 2017-04-01:
Hey,

Before run 'make' run this : ./configure openssl_CFLAGS=-I/usr/local/opt/openssl/include/openssl/ openssl_LIBS=-L/usr/local/opt/openssl/lib/

Before running this you should install openssl with homebrew with the command brew install openssl

nikias, 2017-04-26:
This is probably due to the use of OpenSSL 1.1.0. I just merged pull request #494 which addresses the issue. If the problem persists, feel free to reopen the ticket.

closed 2017-04-26 by nikias