libimobiledevice/libimobiledevice #445: ios10.2 can't mount with Ubuntu / Linux

eura665, 2016-12-30:
Like to inform you that libimobiledevice to Linux / Ubuntu doesn't work since the upgrade to iOS10 in September. you revived a error cod 5,16 …… can’t mount so you can’t access you iphone through the usb cable. Martin Salbaba make a ppa that work on iOS 10 and 10.1 but now with the 10.2  it doesn't work ;-( https://launchpad.net/~martin-salbaba/+archive/ubuntu/ppa+libimobiledevice.
It’s very annoying that you can’t access your photos. Hope some one can make it work again

jwhitbeck, 2017-01-10:
I also have this issue. I would be happy to help debug.

Mintyjack, 2017-01-11:
Hi, just to add I have this problem and happy to help debug.

0ki, 2017-01-20:
My libimobiledevice version is 1.2.0+git20161018-3salbabix27xenial and works nicely with iOS9 and below.

For iOS10 it only show the "Apps" folder. 
If you don't even have that, please do:

> idevicepair unpair
> idevicepair pair

Do the last one multiple times until successful (follow the hints on screen)
Finally reboot your computer.

Now, you will not have access to your media still. It must be **a simple bug** because it can be _**solved**_ with:

> mkdir ~/Desktop/iOSmount
> ifuse ~/Desktop/iOSmount

Note that you will need root permissions to unmount this:

> sudo umount ~/Desktop/iOSmount

Mintyjack, 2017-01-21:
Thanks. I'm on 1.2.0+dfsg-3~ubuntu0.2. When I try the above commands I get:
`idevice:  command not found`

0ki, 2017-01-21:
Remove the extra space after idevice.

Mintyjack, 2017-01-22:
Aha .. just being dumb there. Can unpair/pair but still no joy, I can't reboot with the phone plugged into  USB, but I have seen that before so probably not related. Thanks anyway.

samrocketman, 2017-01-22:
This issue has already been fixed.  See https://github.com/libimobiledevice/libimobiledevice/issues/413

jritzerfeld, 2017-01-22:
I do not think that this is the same problem as #413 for all commenters. At least @0ki and me can use ifuse to connect.

samrocketman, 2017-01-22:
Have you tried building from master and replicating?  In #413 I explain how I tested from master for my issue.

https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

Could you try from that?

jritzerfeld, 2017-01-22:
I use openSUSE and a friendly co-member promptly created a package of master (45fda81) for testing and it does not help. However, it might be a libgphoto2 problem since "gphoto2 --list-files" sometimes work and at other times it does not while throwing the error message "PTP Invalid Storage ID [... ]Error (-1: 'Unspecified error')".

samrocketman, 2017-01-22:
I've not used gphoto2 before.  I typically just mount using `ifuse` and the visit the DCIM folder for photos.

jritzerfeld, 2017-01-22:
Ah, okay. Mounting the DCIM folder works for me and @0ki.

0ki, 2017-01-23:
Mounting the DCIM parent folder actually...
Anyway, something is wrong as this requires manual mounting with iOS10, but not with iOS8.

samrocketman, 2017-01-23:
@0ki in your case, https://bugs.launchpad.net/ubuntu/+source/libimobiledevice/+bug/1623666/comments/28 may help.

jraah55, 2017-02-02:
[libplist-master 2.zip](https://github.com/libimobiledevice/libimobiledevice/files/748372/libplist-master.2.zip)

jritzerfeld, 2017-02-09:
I asked the gphoto2 project for help by opening an issue there: [iOS 10.2 PTP Invalid Storage ID Error (-1: 'Unspecified error') ](https://github.com/gphoto/libgphoto2/issues/132)
They kindly analyzed the gphoto2 debug logs and the problem occurs when iOS 10.2 reports 0x00000000 as the PTP storage id and this id is invalid. So, there is nothing they can do.

JR1994, 2017-02-09:
I am having similar issues. I am on Debian Sid and cannot access my Documents folder(get a -3 error) for my iPhone, Rhythmbox cannot see my music, and Shotwell cannot access all my photos. I tried unpairing and pairing from the terminal with no luck.

ObsidianParis, 2017-03-10:
Hey folks, people who can't wait may want to check out https://github.com/gphoto/libgphoto2/issues/132 as indicated by @jritzerfeld and pull the today's commits. We obtained good results this afternoon. :-)

It's probably not a definitive version and we're still waiting for iOS 10.3 to come to see if it changes something, but I guess some feedback would be appreciated.

0ki, 2018-01-26:
Any updates on this?

ObsidianParis, 2018-01-28:
Well, the other bug entry has been closed and I personally haven't experienced such another issue recently (on the other hand, I think the last time I dumped my latest pictures was this summer). So the library itself should be working, but you need to check what version is currently pushed on your distribution.

For information, I use a Fedora 25, which was still the current one at the time we fixed the bug. The new version of the library was planed for the next versions but never ported to this one (because the maintainer was too busy, thanks to him anyway).

Try to perform the regular updates on you distribution if (for some reason) it hasn't been done. Otherwise, try to get the last version of the lib, then compile/install it manually. It should work.

Good luck.

christophermlne, 2018-05-20:
Doesn't work at all on Manjaro.