libimobiledevice/libimobiledevice #447: Profiles not backing up just right?

Mradr, 2017-01-02:
Hey guys- not 100% sure but I think libimobiledevice might not be backing or restoring just right with 10.2.

We are installing a MDM profile to the device - but once I do a backup and restore - it seems that the MDM profile gets mess up. Not doing anything fancy, just back up - then restore causes the profile to corrupt.

nikias, 2017-01-03:
Have you made sure the backup is encrypted? Maybe some information is only stored (i.e. backed up) when encrypted.

Mradr, 2017-01-03:
The back is not encrypted, but it might be a thing from Apple now. We need a way to access these files even encrypted - we have a work around for now - but can we start seeing something soon from libimobiledevice to access such files?

DaKaZ, 2017-01-26:
This is actually a bug in iOS 10.2, we opened a bug report with apple and I'd encourage you to do the same.