libimobiledevice/libimobiledevice #448: Continuously getting "Failed to start AFC service 'com.apple.mobile.house_arrest' on the device" error on arch linux while mounting iPhone.

ajinkyajoshi111, 2017-01-03:
Hi, 

I'm using arch linux and while mounting, I'm gatting "Failed to start AFC service 'com.apple.mobile.house_arrest' on the device" continuously.

Also please mention where can I find the updated libimobile packages for ARCH linux.

sebkraemer, 2020-12-01:
Possible duplicate of #9.
For me, IPhone reboot solved the issue of such a failing ifuse mount.

I would still hope that there's something libimobiledevice can do about it on its end.

nikias, 2020-12-01:
There's a service limit for house arrest. It will fail after 5 instances are started; have you checked that you stop/unmount the mounts properly?

sebkraemer, 2020-12-01:
I had another mount open via gvfs-afc, I believe.
Other than that, I don't know. I can't rule out the possibility that there were instances running from weeks ago when I learned about libimobiledevice and tried it out. It's the first time today though that I saw this error and that I tried ` ifuse --document`  mount option.
I'll keep an eye on it. This time at least, the manual unmount was successful.