libimobiledevice/libimobiledevice #454: Getting IP of iPhone

StevieSun, 2017-01-11:
Hi,
I am able to get the wifi mac address with ideviceinfo.
I was wondering if i can get the ip of the device with its UDID.
Any help is highly appreciated.

pavanbachu0604, 2017-04-05:
Any luck ? Even I am looking for solution for that ?

StevieSun, 2017-04-13:
@pavanbachu0604 Hi. I was only able to use arp commands and successfully get the ip of the devices when in same network. But it has its own challenges. You need to check if it suits your needs. And if you find an alternative please post it here. Thanks.!

Archerlly, 2021-02-28:
> use arp commands and successfully get the ip of the devices

how to get the ip of the devices with arp commands？

NirBY, 2022-02-14:
I try using (when imac and iphone on same wifi):
```
arp -i en1 -a | grep $(ideviceinfo -k EthernetAddress)
```
"en1" is my imac's wifi interface, got it using:
```
networksetup -listallhardwareports
```
If anyone manage solving this one out please update.