libimobiledevice/libimobiledevice #456: New release?

anatoli26, 2017-01-21:
Please release a new version and notify the distributions to include it, the latest was published 2 years ago, a lot of things are not working in stock distro installations (v1.2.0).

TheElectronWill, 2017-01-23:
This has already been asked multiple times (but there is still no official release :| ).
If you use Fedora, Ubuntu or a derivative, I've put some useful links for you in the comments of issue #380

anatoli26, 2017-01-25:
@TheElectronWill, thanks for the information and the links. I've just checked all related issues asking for a new release, but haven't found any answer from the maintainers. At the same time I see that the project has recent updates, so it's not abandoned. Was there any explanation for no release for so long?

songhaonangit, 2017-02-08:
Please release a new version

reaniwen, 2017-03-31:
I think the best solution is to clone the newest version code and compile it locally.

anatoli26, 2017-04-01:
Sure, but it's not manageable on more that a couple of computers. Does anybody know why the maintainers don't release a new version?

Mintyjack, 2017-04-01:
There are increasing numbers of people like me who migrated from Windows and are not that tech savvy so can't compile their own 'release'. It's a shame, mint 18 was working fine for me till IOS 10.2. I want to support open source but issues like this make it difficult for 'ordinary Joe' type use.

I considered offering a bounty for a ppa of the latest version, but it's a sticking plaster, what happens when IOS 10.3 or whatever comes along?

LinuxDonald, 2017-04-01:
@TheElectronWill maybe add it to the wiki here? Maybe then we could add more script for making own packages?

TamimiGitHub, 2017-04-07:
@FunkyM Do you guys have any plans to release a new version? As of right now, there are a couple of fixed that came in after 1.2 and we have been using `brew install libimobiledevice --HEAD` to make sure fixes are in. It will be great if we can get an update on this release.
Thanks

samrocketman, 2017-04-26:
For those who can't wait for a release, I've published a guide of how I built libimobiledevice for myself.  Perhaps you may benefit.  https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

franklinyu, 2017-11-01:
Duplicate of #380

FunkyM, 2020-06-15:
Closing as version [1.3.0](/libimobiledevice/libimobiledevice/releases/tag/1.3.0) has been released.

closed 2020-06-15 by FunkyM