libimobiledevice/libimobiledevice #458: idevicediagnostics obliterator

Dantee296, 2017-01-23:
Please help me with  obliterate iPhone device via idevice diagnostics
I know that we can do obliteration via idevice diagnostics but dontt know how tried every thing but it doesn't work

This is how I m sending plist]
Var name = “Obliterator”;

plist_dict_set_item(dict, "Request", plist_new_string(name));
   
	plist_dict_set_item(dict, "DataPartitionOnly", plist_new_bool(1));
	plist_dict_set_item(dict, "ObliterationType", plist_new_string("Wipe"));
	plist_dict_set_item(dict, "ObliterateDataPartition", plist_new_bool(1));
	plist_dict_set_item(dict, "DisplayProgressBar", plist_new_bool(1));
	plist_dict_set_item(dict, "SkipDataObliteration", plist_new_bool(0));
	plist_dict_set_item(dict, "ObliterationMessage", plist_new_bool(1));

ret = diagnostics_relay_send(client, dict);
	plist_free(dict);
	dict = NULL;

	ret = diagnostics_relay_receive(client, &dict);
	if (!dict) {
		return DIAGNOSTICS_RELAY_E_PLIST_ERROR;
	}

	int check = diagnostics_relay_check_result(dict);

I saw working of Obliterator in ilibmoibldevice
https://cgit.sukimashita.com/libimobiledevice.git/commit/?id=8b61d177dbfa5c12d582bd58df8243b6d21571b8

but in latest files i think they have removed it.
kindly someone help me with this
Thanks

Dantee296, 2017-01-27:
Some One Help please ????

Dantee296, 2017-01-27:
@FunkyM  @nikias  Guys Please Help me out with this .
i request 👍

Dantee296, 2017-01-27:
its very very important for me even a little hint or way of doing it would help
application on Windows like "iMazing" They already doing it.
i reverse engineered their app and they were calling function
ObliterateAsync();
@nikias @FunkyM

closed 2017-01-27 by Dantee296

nikias, 2017-01-27:
Why did you delete all the comments? Did it work now or not?