libimobiledevice/libimobiledevice #460: idevicebackup2 ErrorCode 105: Insufficient free disk space on device to restore

DaKaZ, 2017-01-25:
I sure this is my user error and not a code issue, but I cannot find the answer anywhere.  I have a large iPad Mini, it has 77GB of "stuff" and is a 128GB unit.  The General->Storage screen shows Used: 77.55 GB and Free: 44.12 GB.   I am getting this error:

```
ErrorCode 105: Insufficient free disk space on device to restore (44526002176 < 51895967372) (MBErrorDomain/105)
```
I am trying to restore a full backup using the command:
```
idevicebackup2 restore --system --settings --remove --reboot ./
```

I am on a MacOS Sierra machine using binaries compiled today ;)  Backups and Restores of other devices are working fine, it only seems to be "full" devices.  Is there a way to clear space on the device before restoring?  What is the proper way to restore this device?

Thanks!

nikias, 2017-01-26:
The error comes from the device itself, not from idevicebackup2. Delete some of the stuff (around 7gb) and try to restore again.

DaKaZ, 2017-01-26:
Thanks @nikias - is this the same problem iTunes would have then?  I am trying to figure out a way to do it programmatically- any ideas? Maybe we can remove a bunch of apps before starting the restore?

nikias, 2017-01-26:
iTunes would have the same issue, of course. Removing apps could help, however - at least by default - removing them through installation_proxy would not remove the app data ("documents"). Also restoring the backup alone would probably not restore the apps automatically. There is an open ticket for the app restore process, there is something needed to complete it.
I would say it depends on what is taking up the space. For example the camera roll is part of the backup and will be restored too, and it is accessible via AFC. If photos and videos are taking up most of the space you can programmatically delete the media files from the DCIM directory.

DaKaZ, 2017-01-26:
Hmmm, this is a more serious problem than I expected.  I appreciate the background info @nikias - is there a tool in the imobiledevice library that you'd recommend to be able to look the storage info?

samrocketman, 2017-01-26:
You could mount your device using ifuse.

```
mkdir -p ~/usr/mnt
idevicepair pair
ifuse ~/usr/mnt/
```

The DCIM folder contains pictures and video.  That is typically the largest space hog on an iPhone.

DaKaZ, 2017-01-26:
@nikias I think I figured it out but have one more question if you don't mind. I have a full backup from libimobiledevice so I think I can figure out which apps are the largest from there.  But the bug you mentioned is restoring previously purchased apps I think, if I have the full backup, the apps in that backup will get restored, right???

nikias, 2017-01-26:
I am not 100% sure as I never tested this fully. This is the issue I was referring to: https://github.com/libimobiledevice/libimobiledevice/issues/343
I never got a chance to analyze the dump.

nikias, 2017-01-26:
Please understand. Apps are NOT part of the backup, just the data (i.e. files) created by apps. So after the restore if you don't have the app installed and install it later, the app data will be automagically restored by the device (they are kept in a 'Safe Harbor').

closed 2017-01-26 by DaKaZ