libimobiledevice/libimobiledevice #461: (Could not perform backup protocol version exchange, error code -6) Visual c++ 2015 Windows

Dantee296, 2017-01-30:
Hello, i am using idevicebackup2 and continuously facing this error no matter what it do
i am using visual studio ++ 2015 windows 10
kindly let me know what to do , in order to fix this error
Every thing is Latest Every thing is working with ios 10+
Just this project having problem with version exchange 
Many Thanks,

Dantee296, 2017-01-30:
@nikias @FunkyM Please Help me with this

closed 2017-01-31 by Dantee296

qmfrederik, 2017-08-18:
See #441

Dantee296, 2017-08-18:
i have fixed and closed it thanks :) @qmfrederik

A5kar, 2023-09-18:
Hi @Dantee296, I am having the same issue. I am on macOS and installed libimobiledevice with Gentoo Portage. Dependences seem satisfied.

    $ idevicepair validate
    SUCCESS: Validated pairing with device XXXXXXXX-XXXXXXXXXXXX

    $ idevicebackup2 backup ./backup
    Backup directory is "./backup"
    Started "com.apple.mobilebackup2" service on port 54457.
    Could not perform backup protocol version exchange, error code -1

Any pointers as to what I could be missing?