libimobiledevice/libimobiledevice #462: Unable to make libimobiledevice in mingw after updating Development Packages

rcmpayne, 2017-01-30:
Ive been building libimobiledevice for a year now using windows just fine (--without-cython). Today i updated the packages for the following to the latest version so I can attempt to build 1.2.1

autoconf-2.69 
libplist-master
libusb-master
libusbmuxd-master

All the above packages was able to ./configure, make and make install without issue but not when i try to make libimobiledevice i fail.

```
$ ./autogen.sh --without-cython
Configuration for libimobiledevice 1.2.1:
-------------------------------------------

  Install prefix: .........: /usr/local
  Debug code ..............: no
  Python bindings .........: no
  SSL support backend .....: OpenSSL

  Now type 'make' to build libimobiledevice 1.2.1,
  and then 'make install' for installation.
```
```
$ make v=1
C:/MinGW/bin/make  all-recursive
make[1]: Entering directory `c:/1.30.2017_Code/libimobiledevice-master'
Making all in common
make[2]: Entering directory `c:/1.30.2017_Code/libimobiledevice-master/common'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `c:/1.30.2017_Code/libimobiledevice-master/common'

Making all in src
make[2]: Entering directory `c:/1.30.2017_Code/libimobiledevice-master/src'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `c:/1.30.2017_Code/libimobiledevice-master/src'

Making all in include
make[2]: Entering directory `c:/1.30.2017_Code/libimobiledevice-master/include'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `c:/1.30.2017_Code/libimobiledevice-master/include'

Making all in tools
make[2]: Entering directory `c:/1.30.2017_Code/libimobiledevice-master/tools'
C:/MinGW/msys/1.0/bin/sh.exe ../libtool  --tag=CC   --mode=link gcc -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden    -IC:/MinGW/msys/1.0/local/include   -IC:/MinGW/msys/1.0/local/include -IC:/MinGW/msys/1.0/local/include/libxml2   -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES=1 -g -O2    -LC:/MinGW/msys/1.0/local/lib -lssl -lcrypto   -LC:/MinGW/msys/1.0/local/lib -lplist    -o idevice_id.exe idevice_id-idevice_id.o ../src/libimobiledevice.la
libtool: link: gcc -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -IC:/MinGW/msys/1.0/local/include -IC:/MinGW/msys/1.0/local/include -IC:/MinGW/msys/1.0/local/include/libxml2 -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES=1 -g -O2 -o .libs/idevice_id.exe idevice_id-idevice_id.o  
-LC:/MinGW/msys/1.0/local/lib 
-L=C:/MinGW/msys/1.0/local/lib ../src/.libs/libimobiledevice.a 
-L==C:/MinGW/msys/1.0/local/lib 
-lssl -lcrypto /usr/local/lib/libusbmuxd.a /usr/local/lib/libplist.dll.a /usr/local/lib/libxml2.dll.a /usr/local/lib/libz.dll.a /usr/local/lib/libiconv.dll.a 
-lole32 
-lws2_32 
-lgdi32 
-L/usr/local/lib

../src/.libs/libimobiledevice.a(idevice.o): In function `internal_idevice_deinit':
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:96: undefined reference to `mutex_destroy'
../src/.libs/libimobiledevice.a(idevice.o): In function `internal_idevice_init':
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:79: undefined reference to `mutex_init'
../src/.libs/libimobiledevice.a(idevice.o): In function `DllMain@12':
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:122: undefined reference to `thread_once'
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:125: undefined reference to `thread_once'
../src/.libs/libimobiledevice.a(idevice.o): In function `idevice_connection_enable_ssl':
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:683: undefined reference to `userpref_read_pair_record'
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:693: undefined reference to `pair_record_import_crt_with_name'
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:694: undefined reference to `pair_record_import_key_with_name'
../src/.libs/libimobiledevice.a(idevice.o): In function `locking_function':
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:60: undefined reference to `mutex_unlock'
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:58: undefined reference to `mutex_lock'
../src/.libs/libimobiledevice.a(idevice.o): In function `idevice_set_debug_level':
c:\1.30.2017_Code\libimobiledevice-master\src/idevice.c:227: undefined reference to `internal_set_debug_level'
[... 80 lines cut ...]
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:1184: undefined reference to `socket_close'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `connect_usbmuxd_socket':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:126: undefined reference to `socket_connect'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `usbmuxd_read_pair_record':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:1232: undefined reference to `socket_close'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `connect_usbmuxd_socket':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:126: undefined reference to `socket_connect'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `usbmuxd_save_pair_record':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:1271: undefined reference to `socket_close'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `connect_usbmuxd_socket':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:126: undefined reference to `socket_connect'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `usbmuxd_delete_pair_record':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:1308: undefined reference to `socket_close'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `usbmuxd_disconnect':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:1104: undefined reference to `socket_close'
C:/MinGW/msys/1.0/local/lib/libusbmuxd.a(libusbmuxd.o): In function `libusbmuxd_set_debug_level':
c:\1.30.2017_Code\libusbmuxd-master\libusbmuxd-master\src/libusbmuxd.c:1324: undefined reference to `socket_set_verbose'

collect2.exe: error: ld returned 1 exit status
make[2]: *** [idevice_id.exe] Error 1
make[2]: Leaving directory `c:/1.30.2017_Code/libimobiledevice-master/tools'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `c:/1.30.2017_Code/libimobiledevice-master'
make: *** [all] Error 2
```

Dantee296, 2017-01-31:
@rcmpayne 
Hy can you please tell me why m facing MOBILEBACKUP2_E_NO_COMMON_VERSION error in idevicebackup2 . every one is deactivate here they dont like to help
please help me if u know why i m facing this error thanks

rcmpayne, 2017-01-31:
Switching to msys2 fixed all my errors except ideviceinstaller which i will post a bug in that section.

closed 2017-01-31 by rcmpayne