libimobiledevice/libimobiledevice #463: idevicebackup2 Erase iphone

Dantee296, 2017-02-03:
Kindly help me with the final message to obliterate (remote wipe iphone) via idevicebackup2

i m very close to deleting iphone just need a message we send via idevicebackup2 to erase the iphone

Just tell me what is the "Message"

Jan 27 16:16:57 iPhone BackupAgent2[227] <Info>: Received message: <private>.  **[We Sent “Hello” Here]**
Jan 27 16:16:57 iPhone BackupAgent2[227] <Info>: Using protocol version 2.1
Jan 27 16:16:57 iPhone BackupAgent2[227] <Info>: Sending response message: 0 (null) **[“Response OK"]**
Jan 27 16:16:57 iPhone BackupAgent2[227] <Info>: Received message: <private>. **[" WHAT IS The Message Here ????"]** **[i Need this message / Parameters Apple sent here]**
Jan 27 16:16:57 iPhone BackupAgent2[227] <Notice>: Erasing device per request from iT

nikias, 2017-02-03:
Interesting. I never saw that BackupAgent2 is able to do that. I will check it out.

Dantee296, 2017-02-03:
When can u tell me about it ?

Dantee296, 2017-02-03:
Can you tell me some alternative ways of doing obliteration ? @nikias

nikias, 2017-02-03:
The operation is "EraseDevice" and was introduced with iOS 9.0
I never tested it but I can add it to the mobilebackup2 service interface so you can try :)

nikias, 2017-02-03:
Actually the interface doesn't need adaption. Since I feel adding this operation to the idevicebackup2 to is irritating/dangerous, I will make a new tool for that. Stay put, I'll post it in a few.

nikias, 2017-02-03:
Happy testing:
https://gist.github.com/nikias/de3fd7bbba96a32b63ddfb81d8a1e9db

Dantee296, 2017-02-04:
Thanks Alot So nice of you
am gonna try it :)
Thanks Agains Sir

closed 2017-02-04 by Dantee296