libimobiledevice/libimobiledevice #464: Send local notification via idevicenotificationproxy

nctusdk, 2017-02-03:
According to its name, it seems like I can use it to push some notification and make them display on the iPhone as normal local/remote notification. Is that correct? If so, please give an example or two about how to use the tool. I tried many times with no success. 

I ran something like 
```
./idevicenotificationproxy -d post hello
```
It just completed silently. 

However I can confirm other tools works fine on my iOS 10 device including the ifuse tool (which is awesome by the way)

ferrazzuk, 2018-02-20:
Any news on this?

mexmer, 2018-02-20:
you can only send set of preselected messages trough notification proxy, or you can register your observer for notification from device from preexisting set.

for example itunesbackup sends notification to phone, that it's starting backup, and also whent it's done. on the other hand, you can also register you own observer, to get notification, when user changes something on springboard.

AFAIK there is no list of all existing notifications, but some notifications you might find in sourcecode.

iosdec, 2019-04-23:
@mexmer is correct, you can't really use this to display visual notifications. If you're still interested in using this, you probably want to delve into CFNotificationCenter: https://developer.apple.com/documentation/corefoundation/cfnotificationcenter?language=objc

LevyTheDevy, 2024-01-27:
Are there any notifications I can `POST` to wake the device from sleep state? 

If so please provide an example?