libimobiledevice/libimobiledevice #469: application sizes

daniel-bluesea, 2017-02-15:
iTunes displays the application sizes (either by get info for each app or at the device's app list),
can it be done using libimobiledevice?

Kanong017, 2017-02-17:
of course can

I simply said the order to get

**Instproxy_client_options_new
Instproxy_client_options_add
Plist_array_append_item (return_attributes, plist_new_string ("StaticDiskUsage"))
Plist_array_append_item (return_attributes, plist_new_string ("DynamicDiskUsage"))
Plist_dict_set_item (opts, "ReturnAttributes", return_attributes)**

Next
Instproxy_browse

daniel-bluesea, 2017-02-20:
i missed that part at the ideviceinstaller (i had some trouble to compile and run its last versions).
thanks!

closed 2017-02-20 by daniel-bluesea