libimobiledevice/libimobiledevice #471: Battery Serial number

nemecence90, 2017-02-22:
Is there any way to get battery serial number through libimobiledevice? I checked the iOS logs, ideviceinfo and mobilegestalt domains too, but to no luck.

Kanong017, 2017-02-23:
diagnostics_relay_query_ioregistry_entry  -> IOPMPowerSource

nemecence90, 2017-02-23:
@Kanong017 Do i have to write a code around it?

I tried:
`idevicediagnostics ioreg IOPMPowerSource`

It gives me:
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <dict>
> 	<key>IORegistry</key>
> 	<dict>
> 	</dict>
> </dict>
> </plist>

Kanong017, 2017-02-23:
First of all 
Open the diagnostics service
Then
Diagnostics_relay_query_ioregistry_entry (diagnostics_service, null, "IOPMPowerSource", &return)
then
Plist_to_xml (return, &plist_data, &plist_size)

The result you want is in plist_data

nemecence90, 2017-02-23:
Thanks for the response @Kanong017.

I will check and let you know.

nemecence90, 2017-02-23:
Hi @Kanong017: Can you explain me the steps to start diagnostics? I have not used code to test libimobiledevice.

nemecence90, 2017-02-23:
I think if the battery serial number isn't received by idevicediagnostics command we couldn't get it through code too, coz internally libimobiledevice must be using the same code.

Kanong017, 2017-02-23:
idevicediagnostics can't get battery serial number

nemecence90, 2017-02-23:
Thats what i thought! But is there any other way or method?

dnsBlah, 2017-10-10:
I know that 3-4 years ago the ideviceinfo did provide Battery Serialnumber, however this was only the KGB SerialNumber.
This is hardcoded in the device, and is not being rewritten after a Battery replacement and/or restore.
I hope this comes back because its really useful for some people! Maybe a display serial as well ? ;-)

nemecence90, 2017-11-17:
Hahah. sure would be useful. Thanks for the update though, it was informative.

closed 2017-11-17 by nemecence90