libimobiledevice/libimobiledevice #477: Prepare / Skip Activation Screen , via iLibmobildevice

Dantee296, 2017-03-14:
[image]

# Hy guys Kindly Check the Attached image
i m doing it programmeticly using cfg-util (Apple utility comes with apple configurator) 
i want to do the same operation using iLibmobildevice 
if any one can help me with , Service name , parameters  any help / hint would be help full,
i have looked in idevicesyslog , but found nothing

Thanks in Advance...
@nikias  looking forward for ur responce :) Cheers

Dantee296, 2017-03-14:
@nikias any help on this :)

crummy, 2017-03-29:
You managed to skip the preparation screen with cfgutil? Can you provide instructions on this? I've been looking for such a solution for a while: http://apple.stackexchange.com/questions/259821/restore-ios-default-settings-without-user-interaction

Dantee296, 2017-04-11:
@crummy  yes we are doing it using 
https://gist.github.com/FredCox3/0617eba0818a3ba957b7
but i wanna implement my own that why i wanted to ask @nikias

crummy, 2017-04-11:
In my experience with cfgutil there are some screens that refuse to skip (I used `--skip-all` and some, like wifi and the "welcome to iPhone screen", where still present).

I was hoping for a solution that would allow me to, entirely automated, instruct a device to reset to a fresh (or prepared) state where it ends up on the home screen, ready to be interacted with.

jackczhang, 2017-06-07:
@Dantee296 @crummy Do U get the solution in windows?

crummy, 2017-06-07:
No, I have not found any way to skip all initial setup screens on an iOS device.

jackczhang, 2017-06-08:
Is there some tool like cfgutil in windows?

closed 2017-06-08 by Dantee296

mexmer, 2017-06-08:
Nope, apple had such tool, but they discontinued it and doesn't work from ios6 onwards