libimobiledevice/libimobiledevice #483: idevicebackup vs imazing

rcmpayne, 2017-03-27:
how is imazing 2.x able to access a backup file form iOS 10 on Windows and view the log files from a specific app "AppDomain-com.myapp -> Library -> logs" and yet idevicebackup on Windows cant? are they doing something different that the libimobiledevice's does not support?

nikias, 2017-03-28:
I don't understand the issue here. Is idevicebackup2 actually creating a backup for you? If it is, then this is not a bug. You still have to access the backup file on your own; this is not the purpose of idevicebackup2 (which is creating and restoring backups).

rcmpayne, 2017-03-29:
Yes, idevicebackup2 is creating the files but i cant seem to extract them to view log files from them which is a part of unback right? are you aware of anyother CLI way to get app log files?