libimobiledevice/libimobiledevice #486: Error when I try to connect my iphone to ubuntu 16.04 64 bit LTS

Battant, 2017-03-29:
Hello,

Here is my configurations :+1: 

ios 9.3.5
iphone 6 plus

Result : In the shotwell programm, I get this message an I have access to the document folder and photos folder with ubuntu file explorer

Shotwell
Unable to fetch previews from the camera:
Could not claim the USB device (-53)

Ios 10.2.1
iphone 5

Result : In the shotwell programm, I get this message an No access to document folder

Shotwell
Unable to fetch previews from the camera:
Could not claim the USB device (-53)

Could you help me to fix this bug ?

Thanks for your support

Best regards

Battant

dangoodman5425, 2017-06-03:
I'm having the same problem, did you find a fix Battant?

zgoda-mobica, 2017-06-14:
Ubuntu ships with latest **released** libimobiledevice from 2015. For sure it does not work with iOS 10, IDK for 9.

https://github.com/libimobiledevice/libimobiledevice/releases/tag/1.2.0