libimobiledevice/libimobiledevice #487: double udid with iDevice_id -l

phukhanhlee, 2017-03-30:
This error I never seen before.
After install libmobiledevice with brew
iDevice_id -l
b178adb6c68969685b6390a496dc75a7c2e76088
b178adb6c68969685b6390a496dc75a7c2e76088

It return double udid. Seem one for wifi and one for usb. When i plug out usb it still return one with iDevice_id -l -> still return udid. Seem it was cached/

Problem is we can't run ideviceinstaller if it return double.

Environment : Xcode 8.3, jailbreak phone 9.3.1
install with brew
brew uninstall ideviceinstaller
brew uninstall libimobiledevice
brew install --HEAD libimobiledevice
brew link --overwrite libimobiledevice
brew install ideviceinstaller
brew link --overwrite ideviceinstaller

=> please help me.

Kanong017, 2017-05-14:
也许有一个是WiFi链接的?

iamluosm, 2017-06-06:
@phukhanhlee Maybe another is sync from iTunes.

dnsBlah, 2017-10-10:
For some reason there is a 'WiFi-pair'. I have same behavior and it's annoying.

aryalakshmi, 2018-10-08:
Hi,
I am also facing similar issue.
Even after disabling sync from itunes , issue still persists. I could see udid of device listed twice.

https://github.com/libimobiledevice/libusbmuxd/issues/56

 can you please help