libimobiledevice/libimobiledevice #488: Help, i need to find the way to fix it

OsukaS, 2017-04-04:
Hi, i'm working on a Iphone 6s, but when i try tu use any idevice, it send me an error code -5, and the worst problem is that i'm working now o Windows O.S., Please someone help me 
[image]

reaniwen, 2017-04-10:
It seems to be something wrong with your SSL? Have you tried to reinstall Open SSL?

nikias, 2017-05-12:
Another perfect example of how not to report an issue. How did you get the binaries? Did you compile them on your own? If you downloaded them somewhere they might be outdated.
Closing since the report is too unspecific. Nobody can really help you if you don't provide any useful information.

closed 2017-05-12 by nikias