libimobiledevice/libimobiledevice #489: Install enrollment profile on iPad

SamuelYvon, 2017-04-07:
Is there a way to install an enrollment profile on an iPad (or any .mobileconfig) from this library?

I am looking to install a .mobileconfig without requiring user input, since it is required to continue the automated preparation process.

Dantee296, 2017-04-11:
yes i can help you with that, before that let me know can you
prepare iPhone programmeticly

Dantee296, 2017-04-11:
we can help each other i believe
if u know how to prepare iPhone programmeticly

SamuelYvon, 2017-04-11:
That's  essentially what I'm trying to do, but I don't know how

crummy, 2017-05-09:
@SamuelYvon If the device is managed I believe this is possible with Apple's `cfgutil` tool:

`cfgutil --ecid "0xXXXXXXXXXXXXX" --certificate "/path/to/cert.crt" --private-key "/path/to/private.key" install-profile "path/to/mobileconfig.mobileconfig"`

Haven't tested it extensively but including the cert and key should skip the user input requirement.