libimobiledevice/libimobiledevice #490: win32_idevicebackup2 error -6 on restore

seast123, 2017-04-21:
win32_idevicebakup2
https://github.com/libimobiledevice-win32
ios10.0.2
report MOBILEBACKUP2_E_NO_COMMON_VERSION
who help me?thx

nikias, 2017-05-05:
This is an unsupported fork. The main repository code works fine, I just compiled it myself in mingw and tested it with a 10.3.1 device. No issues at all.

seast123, 2017-05-08:
@nikias，did you compile on windows7  or window10? I compiled in vs2015 on windows7.

colereynolds, 2017-05-08:
@nikias - I have successfully compiled libplist in mingw, but when trying to run ./autogen.sh for libusbmuxd, it stops shorts with the following error at the bottom of the output below. Do you know what the problem could be?

`$ ./autogen.sh
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
checking for a BSD-compatible install... /bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking whether gcc and cc understand -c and -o together... yes
checking build system type... i686-pc-mingw32
checking host system type... i686-pc-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... c:/mingw/mingw32/bin/ld.exe
checking if the linker (c:/mingw/mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /c/Perl/c/bin/nm -B
checking the name lister (/c/Perl/c/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -p
checking the maximum length of command line arguments... 8192
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert i686-pc-mingw32 file names to i686-pc-mingw32 format... func_convert_file_msys_to_w32
checking how to convert i686-pc-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for c:/mingw/mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /c/Perl/c/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -DDLL_EXPORT -DPIC
checking if gcc PIC flag -DDLL_EXPORT -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (c:/mingw/mingw32/bin/ld.exe) supports shared libraries... yes
checking whether -lc should be explicitly linked in... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... c:/mingw/mingw32/bin/ld.exe
checking if the linker (c:/mingw/mingw32/bin/ld.exe) is GNU ld... yes
checking whether the g++ linker (c:/mingw/mingw32/bin/ld.exe) supports shared libraries... yes
checking for g++ option to produce PIC... -DDLL_EXPORT -DPIC
checking if g++ PIC flag -DDLL_EXPORT -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (c:/mingw/mingw32/bin/ld.exe) supports shared libraries... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
./configure: line 15195: syntax error near unexpected token `libplist,'
./configure: line 15195: `PKG_CHECK_MODULES(libplist, libplist >= $LIBPLIST_VERSION)'`

nikias, 2017-05-08:
You don't seem to have ```pkg-config``` installed in MinGW. Download this:
http://ftp.gnome.org/pub/gnome/binaries/win32/dependencies/pkg-config_0.26-1_win32.zip
And extract ```pkg-config.exe``` into the appropriate directory; default path is: ```C:\MinGW\msys\1.0\bin```

colereynolds, 2017-05-08:
@nikias - Thanks for that tip! I just tried what you suggested and I'm still getting the same error. Below are the two files under `/local/lib/pkgconfig`:
```
libplist.pc
libplist++.pc
```

Is that all that is needed? I also tried running `make install` for libplist again after setting that environment variable, but that didn't help either.

nikias, 2017-05-08:
You are missing ```pkg.m4```. Get it from here:
https://gnunet.org/svn/Extractor/m4/pkg.m4
And put it to ```C:\MinGW\share\aclocal```.

colereynolds, 2017-05-08:
@nikias - That fixed it! After trying to run `./autogen.sh` again I received some errors from `pkg-config.exe` complaining about a couple of missing dll's. I found `libglib-2.0-0.dll` and `intl.dll` from [here](http://ftp.gnome.org/pub/gnome/binaries/win32) and dropped them in `C:\MinGW\bin`, which got me past that. 

I'm still having trouble getting `./autogen.sh` to succeed though. I'm getting the error below. Do you have any idea why it can't find libplist now? I have confirmed that the `PKG_CONFIG_PATH` environment variable is set to `/local/lib/pkgconfig`, so I'm not sure what else to check.

BTW - Thank you so much for helping me with this. I have very little experience with compiling code, especially on Windows with MinGW. I really appreciate you taking the time to help me out!

```
$ ./autogen.sh
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
checking for a BSD-compatible install... /bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking whether gcc and cc understand -c and -o together... yes
checking build system type... i686-pc-mingw32
checking host system type... i686-pc-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... c:/mingw/mingw32/bin/ld.exe
checking if the linker (c:/mingw/mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /c/Perl/c/bin/nm -B
checking the name lister (/c/Perl/c/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -p
checking the maximum length of command line arguments... 8192
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert i686-pc-mingw32 file names to i686-pc-mingw32 format... func_convert_file_msys_to_w32
checking how to convert i686-pc-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for c:/mingw/mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /c/Perl/c/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
[... 24 lines cut ...]
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... c:/mingw/mingw32/bin/ld.exe
checking if the linker (c:/mingw/mingw32/bin/ld.exe) is GNU ld... yes
checking whether the g++ linker (c:/mingw/mingw32/bin/ld.exe) supports shared libraries... yes
checking for g++ option to produce PIC... -DDLL_EXPORT -DPIC
checking if g++ PIC flag -DDLL_EXPORT -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (c:/mingw/mingw32/bin/ld.exe) supports shared libraries... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libplist... configure: error: Package requirements (libplist >= 1.11) were not met:

No package 'libplist' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libplist_CFLAGS
and libplist_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

nikias, 2017-05-09:
If in mingw ```ls -l $PKG_CONFIG_PATH/libplist.pc``` shows a match then this should just work

colereynolds, 2017-05-09:
That command does return a match, and I just tried running ./autogen.sh again after a reboot and I'm still getting the same error.
```
$ ls -l $PKG_CONFIG_PATH/libplist.pc
-rw-r--r-- 1 Cole Administrators 261 May  8 15:03 /local/lib/pkgconfig/libplist.pc
```

Could we use the alternative workaround noted at the end of that output, or would you recommend not taking that route? I'm willing to try it, I just don't know what value to populate the variables with!
```
Alternatively, you may set the environment variables libplist_CFLAGS
and libplist_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

nikias, 2017-05-09:
I don't know what's wrong with your setup. This:
```
$ pkg-config.exe --print-errors "libplist >= 1.11"
```
Throws no error message for me. Also I get this output:
```
$ pkg-config.exe --cflags "libplist >= 1.11"
-IC:/MinGW/msys/1.0/local/include
```
and
```
$ pkg-config.exe --libs "libplist >= 1.11"
-LC:/MinGW/msys/1.0/local/lib -lplist
```
You can try to define the environment variables:
```
$ export libplist_CFLAGS="-IC:/MinGW/msys/1.0/local/include"
$ export libplist_LIBS="-LC:/MinGW/msys/1.0/local/lib -lplist"
```
Maybe that works for you.

colereynolds, 2017-05-10:
@nikias - That fixed it! Thank you so much!

I got all of the dependencies installed, but I'm getting an error when building libimobiledevice. Would you mind looking and see if you can spot what is wrong? I've tried to make some sense of it, but I'm not familiar enough with it to know where to start.

Thank you so much, again! You are truly one of the most helpful developers I've ever worked with. I really appreciate the time you've spent helping a greenhorn like myself.

`./autogen.sh --without-cython` was successful, as you can see below:
```
$ ./autogen.sh --without-cython
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
checking for a BSD-compatible install... /bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking whether gcc and cc understand -c and -o together... yes
checking build system type... i686-pc-mingw32
checking host system type... i686-pc-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... c:/mingw/mingw32/bin/ld.exe
checking if the linker (c:/mingw/mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /c/Perl/c/bin/nm -B
checking the name lister (/c/Perl/c/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -p
checking the maximum length of command line arguments... 8192
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert i686-pc-mingw32 file names to i686-pc-mingw32 format... func_convert_file_msys_to_w32
checking how to convert i686-pc-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for c:/mingw/mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /c/Perl/c/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
[... 150 lines cut ...]
.libs/idevice.o: In function `idevice_connect':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:294: undefined reference to `usbmuxd_connect'
.libs/idevice.o: In function `internal_connection_send':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:350: undefined reference to `usbmuxd_send'
.libs/idevice.o: In function `internal_connection_receive_timeout':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:397: undefined reference to `usbmuxd_recv_timeout'
.libs/idevice.o: In function `internal_connection_receive':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:450: undefined reference to `usbmuxd_recv'
.libs/idevice.o: In function `idevice_disconnect':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:324: undefined reference to `usbmuxd_disconnect'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_save_pair_record':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:288: undefined reference to `usbmuxd_save_pair_record'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_read_pair_record':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:309: undefined reference to `usbmuxd_read_pair_record'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_delete_pair_record':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:339: undefined reference to `usbmuxd_delete_pair_record'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_read_system_buid':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:185: undefined reference to `usbmuxd_read_buid'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [libimobiledevice.la] Error 1
make[2]: Leaving directory `/Build/libimobiledevice/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/Build/libimobiledevice'
make: *** [all] Error 2
```

colereynolds, 2017-05-11:
@nikias - That fixed it! Thank you so much!

I now have all of the dependencies installed, but I'm getting an error when building libimobiledevice.
I notice it's referencing usbmuxd a lot in the errors. I have not complied that because usbmuxd is dependent on libimobiledevice, and libimobiledevice is dependent on usbmuxd... How did you get around that?

```
Cole@CRDWCOLE01 /Build/libimobiledevice
$ make
make  all-recursive
make[1]: Entering directory `/Build/libimobiledevice'
Making all in common
make[2]: Entering directory `/Build/libimobiledevice/common'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/Build/libimobiledevice/common'
Making all in src
make[2]: Entering directory `/Build/libimobiledevice/src'
  CCLD   libimobiledevice.la
.libs/idevice.o: In function `idevice_event_subscribe':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:182: undefined reference to `usbmuxd_subscribe'
.libs/idevice.o: In function `idevice_event_unsubscribe':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:194: undefined reference to `usbmuxd_unsubscribe'
.libs/idevice.o: In function `idevice_get_device_list':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:209: undefined reference to `usbmuxd_get_device_list'
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:222: undefined reference to `usbmuxd_device_list_free'
.libs/idevice.o: In function `idevice_new':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:253: undefined reference to `usbmuxd_get_device_by_udid'
.libs/idevice.o: In function `idevice_connect':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:294: undefined reference to `usbmuxd_connect'
.libs/idevice.o: In function `internal_connection_send':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:350: undefined reference to `usbmuxd_send'
.libs/idevice.o: In function `internal_connection_receive_timeout':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:397: undefined reference to `usbmuxd_recv_timeout'
.libs/idevice.o: In function `internal_connection_receive':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:450: undefined reference to `usbmuxd_recv'
.libs/idevice.o: In function `idevice_disconnect':
C:\MinGW\msys\1.0\Build\libimobiledevice\src/idevice.c:324: undefined reference to `usbmuxd_disconnect'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_save_pair_record':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:288: undefined reference to `usbmuxd_save_pair_record'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_read_pair_record':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:309: undefined reference to `usbmuxd_read_pair_record'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_delete_pair_record':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:339: undefined reference to `usbmuxd_delete_pair_record'
../common/.libs/libinternalcommon.a(userpref.o): In function `userpref_read_system_buid':
C:\MinGW\msys\1.0\Build\libimobiledevice\common/userpref.c:185: undefined reference to `usbmuxd_read_buid'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [libimobiledevice.la] Error 1
make[2]: Leaving directory `/Build/libimobiledevice/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/Build/libimobiledevice'
make: *** [all] Error 2
```

seast123, 2017-05-12:
@nikias,help me,thx, this is my log
$ make
make  all-recursive
make[1]: 进入目录“/d/code/libusbmuxd”
Making all in common
make[2]: 进入目录“/d/code/libusbmuxd/common”
  CC       collection.lo
  CCLD     libinternalcommon.la
make[2]: 离开目录“/d/code/libusbmuxd/common”
Making all in src
make[2]: 进入目录“/d/code/libusbmuxd/src”
  CC       libusbmuxd.lo
  CCLD     libusbmuxd.la
make[2]: 离开目录“/d/code/libusbmuxd/src”
Making all in include
make[2]: 进入目录“/d/code/libusbmuxd/include”
make[2]: 对“all”无需做任何事。
make[2]: 离开目录“/d/code/libusbmuxd/include”
Making all in tools
make[2]: 进入目录“/d/code/libusbmuxd/tools”
  CCLD     iproxy.exe
./.libs/lt-iproxy.c: In function 'main':
./.libs/lt-iproxy.c:319:16: warning: implicit declaration of function '_spawnv' [-Wimplicit-function-declaration]
   rval = (int) _spawnv (_P_WAIT, lt_argv_zero, (const char * const *) newargz);
                ^~~~~~~
/tmp/ccmjlt3v.o: In function `main':
/d/code/libusbmuxd/tools/./.libs/lt-iproxy.c:319: undefined reference to `_spawnv'
/d/code/libusbmuxd/tools/./.libs/lt-iproxy.c:319:(.text.startup+0x4b4): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `_spawnv'
collect2: error: ld returned 1 exit status
strip: './iproxy.exe': No such file
../libtool: line 11083: ./iproxy.exe: No such file or directory

nikias, 2017-05-12:
@seast123 @colereynolds please guys, this is not a discussion forum. The issues you have is because of something not being set up properly on your system.

@seast123: Something seems wrong with your libtool package. Not idea what.

@colereynolds: It seems like it doesn't know it should link against libusbmuxd. I have no idea why. You can do a a ```make V=1``` to get a verbose output when compiling, showing if it actually passed libusbmuxd as  a library to link against.

These are general build issues and not issues in libimobiledevice. Closing issue.

closed 2017-05-12 by nikias

colereynolds, 2017-05-12:
I understand. Thanks for your help, @nikias!

M3taSapien, 2018-07-10:
Came across this post today after encountering the same issue when attempting run **make** upon successfully completing autogen.sh via :
**`PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --prefix=/usr/local`**

After failing to google trowing for an answer, I bit the bullet and read the whole info for the libtool utility for clarity as to why **libusbmuxd** was playing poltergeist during the attempt to build out **libimobiledevice* * even though it was clearly present within the appropriate library. 

After reading the very thorough, and depressingly long memoirs of libtool, I became aware of two key issues that were causing my issue, and may be the solution to anyone who stumbles across this post at somepoint: 
1- Symbolic links for the libtool objects **.la** do not correctly reference back to the required library system path **[ /usr/lib64/]** where libusbmuxd library is located unless you specifically include that path via **ldconfig** OR by **adding the library** by name or location explicitly to env when running autgen.sh. Failing to do this wont cause autogen.sh to throw an obvious error or fail to complete, but prevents libtool from linking the libiusbmuxd library to the libimobiledevice object created, making the sym link it uses to reference during build stage useless, causing the no such file found error. 
** Resolution ** : ** ldconfig** method. I just created a new file under **[/etc/ld.so.conf.d/libiusbmuxd.conf]** which pointed to the file (not the sym link) in **[/local/lib64]** and then for safe measure, I reinstalled libtool packages (devel one as well) for safe measure, removed old src folder for libimobiledevice package and re-ran autgen.sh -> make -> make install this time without any issues. 

2- If you specified a prefix location when executing autogen.sh as to where to install the built out utility, you need to install the library object found under **./src/libimobiledevice.la** using libtool. I advise you don't change the prefix from /usr/local unless you have no other option.

Hope this helps someone having compiling issues. For reference, I'm on Fedora W27 x64. The solution should work on Debian/Ubuntu based distros with little change, if any.