libimobiledevice/libimobiledevice #491: idevicebackup2 / error code -6 / iOS 10.2.1

colereynolds, 2017-04-24:
Every time I try to back my iPhone up, I get the following error:
Could not perform backup protocol version exchange, error code -6

I'm running this with a build from a week ago

Steps I've already taken:
I have ran idevicepair and re-paired and validated the pairing records.

Attached is the output from idevicebackup2 with the -d parameter as well as a snip of the syslog (using idevicesyslog) from the time of the event. Could you please help me find out what is wrong?

Thank you in advance!

[idevicebackup2_debug_output.txt](https://github.com/libimobiledevice/libimobiledevice/files/953060/idevicebackup2_debug_output.txt)
[idevicebackup2_syslog.txt](https://github.com/libimobiledevice/libimobiledevice/files/953067/idevicebackup2_syslog.txt)

seast123, 2017-04-27:
I am, too。For the same!

seast123, 2017-04-27:
On windows?

colereynolds, 2017-04-27:
Yes. That is correct.

@nikias  - Do you have any ideas?

seast123, 2017-05-05:
@nikias

nikias, 2017-05-05:
It works on macOS with a 10.3.1 device. So it must have to do with how the libraries and tools have been compiled on Windows.

nikias, 2017-05-05:
I tried it on win32 with libraries+tools compiled in mingw. Works fine.

nikias, 2017-05-07:
duplicate of #490

closed 2017-05-07 by nikias