libimobiledevice/libimobiledevice #495: Wifi profile doesn't work with idevicebackup2 on iOS > 10.2.1

cffls, 2017-04-25:
I have a wifi profile installed on the device with os version 10.2.1. The profile enable the device automatically connected to a configured WiFi access point when the WiFi is turned on. The WiFi profile works perfectly when the first time that I installed it. However, after backup the device using idevicebackup2 and restoring the device, the device wasn't able to reconnect to the WiFi. As far as I know, this problem doesn't happen on os version < 10.2.1. Is there anything we can do with idevicebackup2 to fix this problem?

mexmer, 2017-04-26:
was your backup password protected? otherwise wifi passwords are not saved to backup (and therefore also not restored)

cffls, 2017-04-26:
@mexmer I tried with encrypted backup and now the WiFi is working. Thank you!

closed 2017-04-26 by cffls