libimobiledevice/libimobiledevice #497: iOS 10.3.1 Mounts But Won't Pair, Reboots iPhone on Disconnect

malanmurphy, 2017-04-29:
Unfortunately libimobiledevice doesn't appear to be working with 10.3.1. Everything installed correctly and no obvious errors, and lsusb sees the device (although after first connect my system hangs for a few seconds), but idevicepair returns a "no device found" error (details below). More problematic, though, is that every time I unplug the iphone from my system, the iphone immediately reboots. Happens every time. I assume this is something very new (and very bad) with 10.3.1.

user@linux:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial

user@linux:~$ lsusb
Bus 001 Device 004: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6

user@linux:~$ sudo systemctl status usbmuxd
● usbmuxd.service - Socket daemon for the usbmux protocol used by Apple devices
Loaded: loaded (/lib/systemd/system/usbmuxd.service; static; vendor preset: enabled)
Active: active (running) since Sat 2017-04-29 15:44:29 PDT; 2min 2s ago
Docs: man:usbmuxd(8)
Main PID: 1658 (usbmuxd)
Tasks: 2
Memory: 844.0K
CPU: 13ms
CGroup: /system.slice/usbmuxd.service
└─1658 /usr/local/sbin/usbmuxd --user usbmux --systemd

Apr 29 15:44:29 vmngplus systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Apr 29 15:44:29 vmngplus usbmuxd[1658]: [16:13:29.643][3] usbmuxd v1.1.1 starting up
Apr 29 15:44:29 vmngplus usbmuxd[1658]: [16:13:29.647][3] Successfully dropped privileges to 'usbmux'
Apr 29 15:44:29 vmngplus usbmuxd[1658]: [16:13:29.657][3] Could not get old configuration descriptor for device 1-4: -5
Apr 29 15:44:51 vmngplus usbmuxd[1658]: [16:13:51.120][2] Could not get serial number for device 1-4: -7
Apr 29 15:44:51 vmngplus usbmuxd[1658]: [16:13:51.120][3] Initialization complete
Apr 29 15:44:51 vmngplus usbmuxd[1658]: [16:13:51.120][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.

user@linux:~$ idevicepair pair
No device found, is it plugged in?

Output from /var/log/syslog:

Sep 26 17:02:22 linux kernel: [ 208.325808] usb 1-2: new high-speed USB device number 3 using ehci-pci
Sep 26 17:02:22 linux kernel: [ 208.472043] usb 1-2: New USB device found, idVendor=05ac, idProduct=12a8
Sep 26 17:02:22 linux kernel: [ 208.472047] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 26 17:02:22 linux kernel: [ 208.472049] usb 1-2: Product: iPhone
Sep 26 17:02:22 linux kernel: [ 208.472051] usb 1-2: Manufacturer: Apple Inc.
Sep 26 17:02:22 linux kernel: [ 208.472060] usb 1-2: SerialNumber: f0ddxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Sep 26 17:02:23 linux systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.440][3] usbmuxd v1.1.0 starting up
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.441][3] Successfully dropped privileges to 'usbmux'
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.445][3] Could not get old configuration descriptor for device 1-3: -5
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.471][3] Connecting to new device on location 0x10003 as ID 1
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.471][3] Initialization complete
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.471][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Sep 26 17:02:23 linux usbmuxd[2143]: [17:02:23.472][3] Connected to v2.0 device 1 on location 0x10003 with serial number f0ddxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Sep 26 17:02:23 linux kernel: [ 209.584713] ipheth 1-2:4.2: Apple iPhone USB Ethernet device attached

...and when I disconnect the device, output from syslog (and then iPhone hard reboots):

Apr 29 15:57:46 linux kernel: [ 1606.551241] usb 1-2: USB disconnect, device number 4
Apr 29 15:57:46 linux usbmuxd[1658]: [16:26:46.536][3] usbmuxd shutting down
Apr 29 15:57:46 linux usbmuxd[1658]: [16:26:46.637][3] Shutdown complete

samrocketman, 2017-05-01:
I [built it from source](https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc) on Ubuntu 16.04 and the master branches work for me.  The maintainers still need to release it though so it can be packaged by distributions.

pixerit, 2017-06-07:
I'm having having the same issue on iOS 10.3.2, master branch on Ubuntu 16.04. iOS hard reboots.

mrmacete, 2018-03-28:
same problem here. are you on vmware?

arnowillig, 2018-03-28:
I got the same problem with iOS 11.2.6 and Ubuntu (on Parallels VM). I got the feeling this has something to do with the virtual machine. But why the reboot on disconnect?

Some log:
[ 9155.446043] usb 1-6: new high-speed USB device number 13 using ehci-pci
[ 9155.579089] usb 1-6: New USB device found, idVendor=05ac, idProduct=12a8
[ 9155.579091] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9155.579093] usb 1-6: Product: iPhone
[ 9155.579094] usb 1-6: Manufacturer: Apple Inc.
[ 9155.579095] usb 1-6: SerialNumber: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
usbmuxd[22606]: [15:44:55.436][3] usbmuxd v1.1.0 starting up
usbmuxd[22606]: [15:44:55.436][3] Successfully dropped privileges to 'usbmux'
usbmuxd[22606]: [15:44:55.439][3] Could not get old configuration descriptor for device 1-13: -5
usbmuxd[22606]: [15:45:14.712][2] Could not get serial number for device 1-13: -7
usbmuxd[22606]: [15:45:14.712][3] Initialization complete
usbmuxd[22606]: [15:45:14.712][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
usbmuxd[22606]: [15:45:16.717][2] Could not get serial number for device 1-13: -7
usbmuxd[22606]: [15:45:18.721][2] Could not get serial number for device 1-13: -7

mrmacete, 2018-03-28:
it seems for me the problem is it gets a timeout here https://github.com/libimobiledevice/usbmuxd/blob/master/src/usb.c#L327

mrmacete, 2018-03-28:
(you are on a previous version so it manifests itself in a different place, but it may be the same underlying problem)

mrmacete, 2018-03-30:
so, apparently the problem is that for some reason when running in a virtual machine iOS 11 exposes one more USB configuration (5 configurations instead of 4) and usbmuxd is assuming the last configuration is the right one to set here: https://github.com/libimobiledevice/usbmuxd/blob/9dfc1eb993a886532c38553a4eeefcfa79180417/src/usb.c#L415

instead in this case the right one is still `4` even if the descriptor advertises 5 configurations. Trying to set the configuration `5` triggers all the observed problems. 

Apparently apple's implementation successfully uses the 5th if available, but linux kernel fails for some reason (i see a lot of `EHCI: Setup packet too short: 0 bytes` and `EHCI: mismatched control requests` errors in vmware debug log).

mrmacete, 2018-03-30:
also: the same problem can be triggered by `lsusb -v` even if usbmuxd is uninstalled, so this confirms the problem is down into the usb stack

nikias, 2018-03-30:
Thanks for tracking this down!

mieszko4, 2019-04-14:
I have the same problem with iPhone 8 iOS 12.2 and mac OS x 10.14.4.
I am getting something like:
```

mrmacete, 2019-04-14:
i think the iphone exposes that interface because it detects the macos host, but then the linux guest's kernel is failing on it for some reason.

mieszko4, 2019-04-14:
Thanx @mrmacete Just my case is that host is linux and guest is macos (you wrote it other way around) - not sure if it means the same problem.
So if I wanted to find resolution here I believe I would need to fix my **host** linux kernel, right?
I need to educate myself more about how usb signal is being proxied. I thought that vmware player simply passes the usb signals from host to guest and that it has nothing to do with drivers used on the host...

nikias, 2019-04-14:
@mieszko4 see the pull request from @mrmacete: https://github.com/libimobiledevice/usbmuxd/pull/91 I guess I should merge it...

mieszko4, 2019-04-14:
oh nice, thanks @nikias! I will will try to build it from that pr and see if I get any luck with this

mrmacete, 2019-04-14:
yes but @mieszko4, your scenario is actually the opposite if mine. did you try to just disable the usbmuxd service in the host?

mieszko4, 2019-04-14:
yes, you are right @mrmacete.
I did try with usbmuxd stopped and the same issue occured.
VMWare uses `/usr/bin/vmware-usbarbitrator` for assigning usb to guest - I believe it is making the problem (perhaps it also uses usbmuxd code?) - I need to investigate this more.

mieszko4, 2019-04-25:
I still did not have too much time to figure it out.
So far I managed to disable native `/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/Resources/usbmuxd` on macOS x guest and when I connect and disconnect iPhone there are no errors and it does not get rebooted (as expected).

So my idea is to build and run https://github.com/libimobiledevice/usbmuxd/pull/91 on macOS x to see if the same problem happens. But I need to investigate how.

hduarte, 2019-07-16:
> @mieszko4 see the pull request from @mrmacete: [libimobiledevice/usbmuxd#91](https://github.com/libimobiledevice/usbmuxd/pull/91) I guess I should merge it...

Yes please

nikias, 2019-07-16:
I merged PR https://github.com/libimobiledevice/usbmuxd/pull/91 which should solve the reboot on disconnect issue.

mieszko4, 2020-10-19:
Hey guys,

I've finally took some time to play round with this and I've managed to resolve my problem.
`usbmuxd@1.1.1` (which includes the mentioned https://github.com/libimobiledevice/usbmuxd/pull/91) helped me with this.
Below are some details regarding the version

## The solution
Use open source `usbmuxd@1.1.1` on the macOS guest instead of the original usbmuxd. The original seems to have an issue when used as a guest os of vmware (https://github.com/libimobiledevice/libimobiledevice/issues/497#issuecomment-377523829)

Some details:

### vmrare - .vmx file
- `VERSION`: VMware® Workstation 16 Player - 16.0.0 build-16894299

* usb compatibility 2.0
* automaticaly connect - yes
* show all usb input devices - yes
* share bluetooth devices - yes
* nothing of quirks is need to be set up

### guest macOS - Catalina
- `VERSION`: macOS Catalina - 10.15.5

* disable sip https://gist.githubusercontent.com/abelardojarab/0fc2f0af699ad1b42b1ec7937e7b2ee4/raw/9eb33d05a30e5df3d01a947540918ba8eefd813e/disable_sip_vmware.md
* install mac ports (https://www.macports.org/install.php)
* `sudo launchctl unload /Library/Apple/System/Library/LaunchDaemons/com.apple.usbmuxd.plist`
* `sudo port install usbmuxd`
* `sudo usbmuxd --foreground`
* then connect iPhone and let guest to take control

### host linux Mint
- `VERSION`: Linux Mint 20 Cinnamon - Cinamon 4.6.7 - Kernel 5.4.0-51-generic

It was fun to play around - from my side we can close this ticket :)

LaQuang, 2021-04-01:
[image]

I have a Host Ubuntu 20.04 and MacOS Bigsur as Guest. The dialog: The specified device is in use by process:35552 /usr/sbin/usbmuxd on the host operating system. Continuing will detach the device from the Host operating system always popup. 

In Guest MacOS I got: 
 sudo usbmuxd --foreground
Password:
[13:41:28.315][3] usbmuxd v1.1.1 starting up
[13:41:28.317][3] Using libusb 1.0.24
[13:41:28.327][3] Initialization complete
[13:41:48.509][3] Could not get old configuration descriptor for device 2-1: LIBUSB_ERROR_NOT_FOUND
[13:41:50.937][2] Could not set configuration 4 for device 2-1: LIBUSB_ERROR_OTHER
[13:42:10.405][3] Could not get old configuration descriptor for device 2-1: LIBUSB_ERROR_NOT_FOUND
[13:42:13.468][2] Could not set configuration 4 for device 2-1: LIBUSB_ERROR_OTHER

mieszko4, 2021-04-01:
Perhaps you have not disabled sip on the guest?

LaQuang, 2021-04-01:
I did disable SIP on Guest Big Sur already.

andreabellitto, 2021-04-17:
> I did disable SIP on Guest Big Sur already.

The same for me on Fedora 33

mieszko4, 2022-06-12:
May be related to https://github.com/libimobiledevice/usbmuxd/issues/172 (i.e. we need new version of `usbmuxd`)

mieszko4, 2022-06-13:
> May be related to [libimobiledevice/usbmuxd#172](https://github.com/libimobiledevice/usbmuxd/issues/172) (i.e. we need new version of `usbmuxd`)

I've build HEAD of `usbmuxd` (https://github.com/libimobiledevice/usbmuxd/commit/2839789bdb581ede7c331b9b4e07e0d5a89d7d18) on Monterey 12.4 Guest using Quickemu (https://github.com/quickemu-project/quickemu) and I run it with `sudo /usr/local/sbin/usbmuxd --foreground`, I got this error when connecting iOS 15.1.4 iPhone:
```terminal
[09:51:57.058][3] usbmuxd v1.1.2 starting up
[09:51:57.062][3] Using libusb 1.0.26
[09:51:57.068][3] Initialization complete
[09:52:07.956][2] Could not claim interface 1 for device 7-1: LIBUSB_ERROR_NOT_FOUND
```

So yeah, it shows less errors but still no success...