libimobiledevice/libimobiledevice #499: ERROR: Could not connect to lockdownd, error code -21

amzil, 2017-05-04:
I have install libimobiledevice on ubuntu but it not works i have error -21.

Linux rkm1-pc 4.4.0-31-generic #50~14.04.1-Ubuntu SMP Wed Jul 13 01:07:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

What is meant by error -21 ?

> Installation steps : 
> https://gist.github.com/Aaahh/1e354ee60dbe40f64efe7e4f98f720d4

`$ idevice_id -l
6bf631e855214af59d4edd201a21f6d79848b2aa

$ ideviceinfo 
ERROR: Could not connect to lockdownd, error code -21`

also I have another question, how i can get information about icloud activation status from usb by libimobiledevice tools, On ubuntu OS ?

JohnChain, 2017-05-23:
I get this error code too
Anyone can help on this ?

ERROR: Could not connect to lockdownd, error code -21

env:
MacOS 10.12.4

iamluosm, 2017-06-01:
@JohnChain I remember I just Run this command line ```ideviceimagemounter```

closed 2023-10-20 by amzil