libimobiledevice/libimobiledevice #504: idevicerestore - not downloading latest ipsw

Jonathan-Walsh, 2017-05-22:
I am trying to download/restore the latest ipsw to my iPad. The current latest should be 10.3.3 beta 1 (or 10.3.2) but instead it was downloading the 10.2 ipsw. Any thoughts on why this is occurring?

mexmer, 2017-06-05:
do you have beta provisioning profile installed on your device?

nikias, 2017-06-05:
@Jonathan-Walsh I think you are using the ```-l``` option? idevicerestore downloads the ```version.xml``` file from itunes.com to figure out which is the most recent version. Maybe the download or the parsing of the document fails?

nikias, 2017-06-05:
Try to remove ```version.xml``` and re-run idevicerestore.

mexmer, 2017-06-06:
tested just now, and on device with beta profile, and on normal device both downloads 10.3.2.

thought it devices which version to download based on profile (phone with beta profile offers 10.3.3.beta3 for download)