libimobiledevice/libimobiledevice #506: Erase All Content and Settings

Al22lan, 2017-05-24:
Hi! Is it possible to do the Erase All Content and Settings using the functionality of the libimobiledevice, as if it is made when you press a menu function reset of device?

daniel-bluesea, 2017-08-28:
i think not.

you can call an iTunes exe for that.

dnsBlah, 2017-10-10:
It should be possible, Apple Configurator can do that.
Just trigger the Erase function to the device, but with idevicerestore it's not possible.... yet?

daniel-bluesea, 2017-10-16:
you are welcome to RE it, see if this Configurator is addressing a certain service on the device, what plist files are they are sending to each other, etc :)

dnsBlah, 2017-10-24:
Hi Daniel,
I have no idea where to start getting this type of data so this can be developed
I wish I had :( If there is anything I can do to help, it would be fun

yashsharma45, 2018-08-11:
@dnsBlah Hi,

Did you find any way to Trigger Erase function? I was searching for it but i couldn't find it anywhere. Please let me know if you have done that.

Thanks in Advacne....:)

Adderally, 2021-05-31:
3uTools can do the same

ibnbax, 2021-09-02:
if there is solution am ready to pay

yashsharma45, 2021-09-03:
@ibnbax I have Used Raw Code of LibimobileDevice and than compiled and used that. the one in previously compiled build did not work for me.

nikias, 2021-09-03:
@ibnbax DM me on telegram (`@pimskeks`)

ibnbax, 2021-09-04:
> @ibnbax I have Used Raw Code of LibimobileDevice and than compiled and used that. the one in previously compiled build did not work for me.

I'm ready to pay for your solution

ibnbax, 2021-09-04:
> @ibnbax I have Used Raw Code of LibimobileDevice and than compiled and used that. the one in previously compiled build did not work for me.

I'm ready for pay for your solution

Royin, 2021-09-08:
> if there is solution am ready to pay

[image]
use idevicerestore: https://github.com/libimobiledevice/idevicerestore
how much you plan pay for haha~~~~~~~~~

ethanatwork, 2021-09-09:
Check out my pull request: https://github.com/libimobiledevice/libimobiledevice/pull/1029

Download and build that version, and use the command `idevicebackup2 erase`

This starts an erase all content and settings on the device, and as such does not require an IPSW download and it does not update the ios version on the device. If you get an error from that command make sure the device is paired and that the device is activated.

Hope this helps.

Ethan

getcodeSam, 2023-02-09:
hi @ethanatwork
not able to see latest push in which contains erase command

jsonlee1688, 2024-07-22:
> Check out my pull request: #1029
>
> Download and build that version, and use the command `idevicebackup2 erase`
>
> This starts an erase all content and settings on the device, and as such does not [...]

Hi, I have connect to iwatch and excite the cmd: `./idevicebackup2 erase` which returned `Backup directory is ".this_folder_is_not_present_on_purpose"
ERROR: Could not start service com.apple.mobilebackup2.` for me, how to fix this issue?

mexmer, 2024-07-22:
there is no mobilebackup2 service on apple watch. as for erase command, no idea, since watches erase automatically, when you pair them with new phone, or you can do it from settings in watch ui. it might be possible to delete them in recovery mode (just idea, have not tested it)

jsonlee1688, 2024-07-22:
> there is no mobilebackup2 service on apple watch. as for erase command, no idea, since watches erase automatically, when you pair them with new phone, or you can do it from settings in watch ui. it might be possible to delete them in recovery mode (just idea, have not tested it)

So you mean when paired with a new phone the watch's user data will be removed entirely automatically? Absolutely setting has a UI which I can erase all the content, but i just want to erase from devicebackup2.

jsonlee1688, 2024-07-22:
Or is there any other cmd for me to erase the watch without manually erase from setting UI? What about idevicerestore?

mexmer, 2024-07-22:
> So you mean when paired with a new phone the watch's user data will be removed entirely automatically? Absolutely setting has a UI which I can erase all the content, but i just want to erase from [...]

as i said in watch (or rather in watchOS) there is no mobilebackup2 service, therefore idevicebakup2 can do nothing with it. as for other methods of erasing content, no idea.

jsonlee1688, 2024-07-22:
OK, thank you for your support! And last will support watch erase one day?  Or it just cannot install the mobilebackup2 service?