libimobiledevice/libimobiledevice #510: iOS 11 ideviceinstaller, ideviceinfo not working

JonGabilondoAngulo, 2017-06-05:
Hi,
In iOS 11 the libimobiledevice commands that need the lockdown service have stopped working.
Do you have the same problem ?
Any solution ?

Thanks

nikias, 2017-06-05:
Haven't tried yet. Is ```ideviceinfo -s``` working? Any debug output with ```-d```?

sohgoh, 2017-06-06:
Now I tried ideviceinfo command using iPhone 7 with iOS 11(15A5278f).

```ideviceinfo -s``` is working, but ```-d``` is not.

```
% ideviceinfo -d
ERROR: Could not connect to lockdownd, error code -3
%
```

nikias, 2017-06-06:
You didn't compile libimobiledevice with debug support. Re-configure with ```--enable-debug-code```.

sohgoh, 2017-06-06:
I retry with ```--enable-debug-code```.

In the following the data of DeviceCertificate, HostCertificate, RootCertificate are deleted and EscrowBag and other ids are modified by me.
```
% ideviceinfo -d
11:53:55 lockdown.c:675 lockdownd_client_new(): device udid: 1bf59ed911dc402c423e18f30ee70fd3fbedc348a
11:53:55 lockdown.c:405 lockdownd_query_type(): called
11:53:55 property_list_service.c:128 internal_plist_send(): sending 284 bytes
11:53:55 service.c:140 service_send(): sending 4 bytes
11:53:55 service.c:140 service_send(): sending 284 bytes
11:53:55 property_list_service.c:133 internal_plist_send(): sent 284 bytes
11:53:55 property_list_service.c:134 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
11:53:55 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
11:53:55 property_list_service.c:202 internal_plist_receive_timeout(): 297 bytes following
11:53:55 property_list_service.c:215 internal_plist_receive_timeout(): received 297 bytes
11:53:55 property_list_service.c:241 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
11:53:55 lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
11:53:55 property_list_service.c:128 internal_plist_send(): sending 5035 bytes
11:53:55 service.c:140 service_send(): sending 4 bytes
11:53:55 service.c:140 service_send(): sending 5035 bytes
11:53:55 property_list_service.c:133 internal_plist_send(): sent 5035 bytes
11:53:55 property_list_service.c:134 internal_plist_send(): printing 5035 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>PairRecord</key>
	<dict>
		<key>DeviceCertificate</key>
		<data>

(deleted)

		</data>
		<key>EscrowBag</key>
		<data>
		KFuRgF3rLcsJ3Fwcu455P8rgZHq5E8d0l8R6g3V42f0=
		</data>
		<key>HostCertificate</key>
		<data>

(deleted)

		</data>
		<key>HostID</key>
		<string>2B0A7912-815A-4944-8731-3E395852052F</string>
		<key>RootCertificate</key>
		<data>

(deleted)

		</data>
		<key>SystemBUID</key>
		<string>A57D15E8-02AE-4298-B74D-56824CEE970F</string>
		<key>WiFiMACAddress</key>
		<string>18:ef:1f:e4:ea:fa</string>
	</dict>
	<key>Request</key>
	<string>ValidatePair</string>
	<key>ProtocolVersion</key>
	<string>2</string>
</dict>
</plist>
11:53:55 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -35 (Undefined error: 0)
11:53:55 service.c:163 service_receive_with_timeout(): could not read data
11:53:55 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
11:53:55 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
ERROR: Could not connect to lockdownd, error code -8
%
```

nikias, 2017-06-06:
Hmm the device just disconnects early as it seems. Apple changed something apparently.

nikias, 2017-06-06:
Can you try to run ```idevicepair unpair``` and paste the output of ``idevicepair -d pair``` ?

rcmpayne, 2017-06-06:
Same issue here on windows with iOS 11 beta. 10.3.3 is working without issue

git clone http://git.libimobiledevice.org/libimobiledevice.git
git pull
./autogen.sh
./configure --without-cython --enable-debug-code --prefix=/c/6.6.2017_debug
Make
Make install

`
C:\6.6.2017_debug>ideviceinfo.exe -s works
C:\6.6.2017_debug>ideviceinfo.exe -d fails 
`
`
C:\6.6.2017_debug>idevicepair.exe unpair
SUCCESS: Unpaired with device 0dc8ad0d68e7fb8b5299c15c971c6f2c4953ba98
`
`
C:\6.6.2017_debug>idevicepair -d pair
14:01:28 lockdown.c:675 lockdownd_client_new(): device udid: 0dc8ad0d68e7fb8b5299c15c971c6f2c4953ba98
14:01:28 lockdown.c:405 lockdownd_query_type(): called
14:01:28 property_list_service.c:128 internal_plist_send(): sending 284 bytes
14:01:28 service.c:140 service_send(): sending 4 bytes
14:01:28 service.c:140 service_send(): sending 284 bytes
14:01:28 property_list_service.c:133 internal_plist_send(): sent 284 bytes
14:01:28 property_list_service.c:134 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicepair</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
14:01:28 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
14:01:28 property_list_service.c:202 internal_plist_receive_timeout(): 297 bytes following
14:01:28 property_list_service.c:215 internal_plist_receive_timeout(): received 297 bytes
14:01:28 property_list_service.c:241 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
14:01:28 lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
14:01:28 property_list_service.c:128 internal_plist_send(): sending 333 bytes
14:01:28 service.c:140 service_send(): sending 4 bytes
14:01:28 service.c:140 service_send(): sending 333 bytes
14:01:28 property_list_service.c:133 internal_plist_send(): sent 333 bytes
14:01:28 property_list_service.c:134 internal_plist_send(): printing 333 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicepair</string>
        <key>Key</key>
        <string>DevicePublicKey</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
14:01:28 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
14:01:28 property_list_service.c:202 internal_plist_receive_timeout(): 914 bytes following
[... 321 lines cut ...]
                LS0K
                </data>
                <key>SystemBUID</key>
                <string>305843672103436834131978564</string>
                <key>HostID</key>
                <string>D966B752-5AEF-08A2-390F-F35DFF0A294F</string>
                <key>EscrowBag</key>
                <data>
                PLO2DixMnY2291+WbnYhFv/XuUV9hAj9HYy/n3B5HAE=
                </data>
                <key>WiFiMACAddress</key>
                <string>f0:24:75:82:a0:53</string>
        </dict>
        <key>Request</key>
        <string>ValidatePair</string>
        <key>ProtocolVersion</key>
        <string>2</string>
</dict>
</plist>
14:04:27 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (No error)
14:04:27 service.c:163 service_receive_with_timeout(): could not read data
14:04:27 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
14:04:27 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
ERROR: Could not connect to lockdownd, error code -8
`

sohgoh, 2017-06-07:
I tried. It seems to be successful.

* idevicepair unpair
```
% idevicepair unpair
SUCCESS: Unpaired with device 1b659ed991dc402c423e18f30ee70fd3fbedc38f
%
```

* idevicepair -d pair
```
% idevicepair -d pair
09:03:41 lockdown.c:675 lockdownd_client_new(): device udid: 1b659ed991dc402c423e18f30ee70fd3fbedc38f
09:03:41 lockdown.c:405 lockdownd_query_type(): called
09:03:41 property_list_service.c:128 internal_plist_send(): sending 284 bytes
09:03:41 service.c:140 service_send(): sending 4 bytes
09:03:41 service.c:140 service_send(): sending 284 bytes
09:03:41 property_list_service.c:133 internal_plist_send(): sent 284 bytes
09:03:41 property_list_service.c:134 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicepair</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
09:03:41 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
09:03:41 property_list_service.c:202 internal_plist_receive_timeout(): 297 bytes following
09:03:41 property_list_service.c:215 internal_plist_receive_timeout(): received 297 bytes
09:03:41 property_list_service.c:241 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
09:03:41 lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
09:03:41 property_list_service.c:128 internal_plist_send(): sending 333 bytes
09:03:41 service.c:140 service_send(): sending 4 bytes
09:03:41 service.c:140 service_send(): sending 333 bytes
09:03:41 property_list_service.c:133 internal_plist_send(): sent 333 bytes
09:03:41 property_list_service.c:134 internal_plist_send(): printing 333 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicepair</string>
        <key>Key</key>
        <string>DevicePublicKey</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
09:03:41 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
09:03:41 property_list_service.c:202 internal_plist_receive_timeout(): 914 bytes following
09:03:41 property_list_service.c:215 internal_plist_receive_timeout(): received 914 bytes
09:03:41 property_list_service.c:241 internal_plist_receive_timeout(): printing 914 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Key</key>
[... 94 lines cut ...]
</dict>
</plist>
09:03:42 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
09:03:42 property_list_service.c:202 internal_plist_receive_timeout(): 316 bytes following
09:03:42 property_list_service.c:215 internal_plist_receive_timeout(): received 316 bytes
09:03:42 property_list_service.c:241 internal_plist_receive_timeout(): printing 316 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>EscrowBag</key>
        <data>
        /AIWwoNXU/VzFKTEeGl9gU0DhuuY/QbkHeVS1N1uYla=
        </data>
        <key>Request</key>
        <string>Pair</string>
</dict>
</plist>
09:03:42 lockdown.c:967 lockdownd_do_pair(): Pair success
09:03:42 lockdown.c:969 lockdownd_do_pair(): internal pairing mode
09:03:42 lockdown.c:978 lockdownd_do_pair(): Saving EscrowBag from response in pair record
09:03:42 lockdown.c:984 lockdownd_do_pair(): Saving WiFiAddress from device in pair record
SUCCESS: Paired with device 1b659ed991dc402c423e18f30ee70fd3fbedc38f
%
```

sohgoh, 2017-06-07:
It seems that response failed at line 730 of lockdown.c and the response is returning with LOCKDOWN_E_ESCROW_LOCKED (-35).
Temporarily commenting out will make ideviceinfo successful.
```
     /* in any case, we need to validate pairing to receive trusted host status */
     ret = lockdownd_validate_pair(client_loc, NULL);
```

nikias, 2017-06-07:
The ValidatePair request was removed from iOS lockdownd, so it doesn't understand it. It will return that the command is invalid.
Commenting that line in lockdown.c out will make it work when the device is already paired, yes. Since we need to keep backwards compatibility I need to work out a solution that works in all cases.

EliyahuStern, 2017-06-07:
How about:
- Add a version double to idevice struct.
- Initialize it inside get_device_info (if it is simple).
- in lockdownd_client_new_with_handshake:
  - if os version is not set, call get_device_info(simple).
  - skip lockdownd_validate_pair call in in case version >= 11.0.

nikias, 2017-06-07:
@EliyahuStern generally yes, just need to make sure the actual pairing works with the open source usbmuxd for a new pairing.
Even for < 11 it actually works without validatepair (since starting a session only works with a correct pairing).

mexmer, 2017-06-07:
@nikias just installed ios 11 beta little while ago, and found that my code is not working (early disconnect after validate pair), so i checked with libimobile and same issue. 
then i decided to look at github, and see you are already working on it.

i still plan to do communication capture (with hardware usb monitor), since itunes for windows seems to be working with ios11 beta. 

they might have something up in the sleeves, that makes seamless communication with old and new devices possible 🐱

nikias, 2017-06-07:
@mexmer as I said, ValidatePair is technically not required even for older devices. At some point IIRC it was required to gain trusted host status (```ideviceinfo -k TrustedHostAttached```), but at least on iOS 5.0.1 it would also be true without performing ValidatePair.

mexmer, 2017-06-07:
i would probly leave it in my app for elder devices, but seems that ios 11 works without it. will need to redo session start handling tho', (i have different workflow that libimobile tools, and some stuff is running in paralel )

i still plan to check itunes communication.

not sending validatepair saves one roundtrip tho' 😄

nikias, 2017-06-08:
Ok I found an even older device with iOS 4.2.1 and this one actually expects ValidatePair. While the oss usbmuxd actually handles all pre-iOS 7 devices separately, there is no change required for a previously unknown device. However if the device is already paired - and this also applies to the official usbmuxd - we still have to call ValidatePair to get trusted host status.

EliyahuStern, 2017-06-08:
Sounds like ValidatePair can be called only as a fallback for failed sessions (maybe after checking the os version).

mexmer, 2017-06-08:
Nope, like nikias said, preios 5 devices require validatepair. While ios11 disconnects you, when you send validaterepair

mexmer, 2017-06-08:
Thought my ipad with ios5 tends to reject session, if i don't use validate pair before secure service start

nikias, 2017-06-08:
@mexmer oh hmm... so if you do ```ideviceinfo -k TrustedHostAttached``` will it show false in that case?

mexmer, 2017-06-08:
is there usbmuxd available with this modification already? i'm talking about my app, that communicates with iDevice directly (trough USB endpoints)

nikias, 2017-06-08:
oh hm. maybe I need to do some more testing then. The open source usbmuxd does not require any changes since for versions before 7 it will call validate pair for a new pairing. For known devices the code change has to be in libimobiledevice to make sure validatepair is called for older devices.

mexmer, 2017-06-09:
I will let you know, if i find out more. For now i "fixed" that in my app by calling validatepair for all pre trust (ios7) devices.

nikias, 2017-06-09:
I came up with this solution:
https://gist.github.com/nikias/b351bf633d62703e0ff4f2fee9628401

mexmer, 2017-06-09:
thanks i will check that.

sohgoh, 2017-06-12:
I tried @nikias 's patch with the following devices and iOS version, ideviceinfo worked in all combinations. (:-)

iPhone 4S, iOS 6.1.2
iPhone 5, iOS 6.1.4
iPhone 6, iOS 10.3.1
iPhone 7, iOS 11 beta

mexmer, 2017-06-13:
looks fine.
iPad 1stGen - iOS 5.1.1
iPhone 2G - iOS 3.1.2

rekyagis, 2017-06-14:
@nikias 's patch works well with following devices / iOS versions:
iPhone 3GS with iOS 4.2.1
iPhone 5c with iOS 7.1.2
iPhone 6 with iOS 8.4.1
iPhone 6 Plus with iOS 9.3.5

Thank you for nice solution.

VasilijSviridov, 2017-06-14:
are you going to make it global ? do I have to update libimobiledevice with brew ? 

Any help ?

JoseCarlosGarcia95, 2017-06-18:
I have tested the patch, and now ifuse is working (tested on iPhone 6 Plus with iOS 11), but in the other hand, if a try to validate  (idevicepair validate) a pairing a get a usbmuxd error, but now I can pair my iPhone without problem.

Maybe the problem came from here?:
+ if (LOCKDOWN_E_SUCCESS == ret) {
+				ret = lockdownd_validate_pair(client_loc, NULL);
+			}
Because your calling the same function very quickly...

nikias, 2017-06-18:
You can't call ```lockdownd_validate_pair``` on iOS 11 because Apple removed the ValidatePair command, it will make the device report an error. I will make a workaround for the idevicepair validate command.
Though actually the idevicepair utility is not really required anymore, since usbmuxd is handling the device pairing.

hugozerken, 2017-06-26:
thx I have use ideviceinstaller successfully!!!
but the configure and make process still have a fill step should avoid
hope someday the installation could be much more smooth in macos
I do search but a lot of answser is based on Ubuntu

truonggiang0710, 2017-06-27:
Hi @nikias ! 
I just modify as your advice. 
I remove all file at /var/lib/lockdown. Then I run to get information, it request Trust. I touch Trust  and get information successful. But
When I unplug and plug phone again. It still request Trust. Can you tell me why, please ?
On iOS 10.3.2, the issue does not appear !

Thanks a lot

stephkenzo, 2017-07-07:
Hi, are there any updates regarding this issue? I see there is a patch but it is not clear to me what is missing for a tool like robovm, which uses libimobiledevice and has this "LOCKDOWN_E_PLIST_ERROR" for now.
I am a basic user, so I might just have missed something ;)
Thanks for the great work anyway.

nikias, 2017-07-07:
@stephkenzo robovm needs to use a newer libimobiledevice with that patch included.

stephkenzo, 2017-07-10:
@nikias Thanks a lot for your answer. Just curious, what remains to be done in this issue to close it?

nikias, 2017-07-10:
I am not 100% satisfied with my patch somehow, so I am trying to come up with something that I like more...

pmnordkvist, 2017-07-10:
Great work @nikias Please let us know if you need help with testing or otherwise.

stephkenzo, 2017-07-11:
Hi @nikias , what are the changes still to be made? Is there any way to help?

maestun, 2017-07-17:
@nikias Could you please provide the source code for the patch, even in its early stage ? Thanks

StefanOltmann, 2017-07-26:
@nikias Can we go with your patch? We'll soon need a new RoboVM version that supports iOS 11.

Haraguroicha, 2017-08-02:
Try this with iOS 11, this use the patch from @nikias

`brew install https://gist.github.com/Haraguroicha/0dee2ee29c7376999178c5392080c16e/raw/libimobiledevice.rb --HEAD --with-ios11`

saurik, 2017-08-12:
@nikias Do you have reason to believe your patch is actively *harmful*? If not, maybe it would be good idea to commit it to the repository, as you can always come up with a better solution later?

nikias, 2017-08-12:
@saurik you are right, it's not harmful so I finally committed it: 5a85432719fb3d18027d528f87d2a44b76fd3e12

closed 2017-08-12 by nikias

truonggiang0710, 2017-08-29:
Hi @nikias,  Do you know about my issue ?
======================================
I just modify as your advice.
I remove all file at /var/lib/lockdown. Then I run to get information, it request Trust. I touch Trust and get information successful. But
When I unplug and plug phone again. It still request Trust. Can you tell me why, please ?
On iOS 10.3.2, the issue does not appear !
=====================================
Thanks a lot

EdgarsAvotins, 2017-11-27:
> Haraguroicha commented on 2 Aug •  edited 
> 
> Try this with iOS 11, this use the patch from @nikias 
> brew install https://gist.github.com/Haraguroicha/0dee2ee29c7376999178c5392080c16e/raw/libimobiledevice.rb --HEAD --with-ios11

Why did this massage have no likes? Was wondering how to apply the patch and this helped. Although I deleted the `--with-ios11` part, because it was failing otherwise. So the commands I used are:
```
brew uninstall libimobiledevice
brew install https://gist.github.com/Haraguroicha/0dee2ee29c7376999178c5392080c16e/raw/libimobiledevice.rb --HEAD
```
Uninstalled libimobiledevice and reinstalled it with the changes from the patch. 
I'm not experienced in this, so if I wrote something dumb, feel free to point it out, so that me and others can learn from my mistakes :)

Haraguroicha, 2017-11-28:
@EdgarsAvotins Because that patch was come from @nikias and only is a workaround, not to confirm it was safe yet.

cnsxxx09, 2024-09-05:
Hi

7 years later and I seem to have the same issue that the patch from [@nikias](https://github.com/nikias) apparently fixed - maybe over the years the patch fix dropped out of the code?

I am using Mac OS Sonoma 14.6.1 (as my Raspberry Pi did not even find the device but the Mac does) and connecting to an original iPad version 1 with a Jailbreak running Cydia.

The error is:
17:47:15 lockdown.c:272 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
ERROR: Could not connect to lockdownd: SSL error (-5)

and when run in debug I see:
17:47:15 idevice.c:642 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -54 (Connection reset by peer)
17:47:15 service.c:167 service_receive_with_timeout(): could not read data
17:47:15 property_list_service.c:229 internal_plist_receive_timeout(): received incomplete packet (611 of 369295618 bytes)

the iPad is connected ok, I can see it is paired, there is no request from the iPad to Trust the Mac however.

Any idea?

thanks!

mexmer, 2024-09-05:
what iOS version is it running? does mac itself see ipad? if mac doesn't see ipad, libimobiledevice will not see it either.

from error **ERROR: Could not connect to lockdownd: SSL error (-5)** i can see two possible causes
1) there was pairing record cached from previous connection, for that should help running 
```idevicepair unpair```
 then either recoonect ipad, or run
```idevice pair```
2) ipad is running rather ancient version of iOS and your version of OSX dropped/disabled support for it.

cnsxxx09, 2024-09-05:
thanks for reply!

iPad is running the latest available version for the original model - 5.1.1 and has the Cydia jailbreak.

I can see that it is paired.
I can unpair and re-pair without problem.

It seems to be supported by MacOSX. I can connect it, I can see it in the Finder, I can restore, sync etc. but if try to copy a .ipa file then I see the icon appear on the iPad, it says waiting, then installing ... and then I get an error - ïTunes Sync .... xxx failed to install'. 
Hence I am trying to use ideviceinstaller instead 

I cannot even run: ideviceinfo -k TrustedHostAttached
without getting an SSL error so this seems (?) to be the problem ....

From all the searching it seems to be

mexmer, 2024-09-06:
ipa files have compatibility, so when you installing with ipa, make sure it's for your ios version or lower.

as for ideviceinfo, sounds weird, if idevicepair works, then ideviceinfo should work also, since it uses same libraries.
can you post output of 
``ideviceinfo -d```

cnsxxx09, 2024-09-06:
Sure, and thanks!, here is the output.

As I mentioned above, SSL connections seems to be where it spits out errors:
11:38:51 idevice.c:1077 idevice_connection_enable_ssl(): Performing SSL handshake
11:38:51 idevice.c:1092 idevice_connection_enable_ssl(): ERROR during SSL handshake: error:0A000152:SSL routines::unsafe legacy renegotiation disabled
11:38:51 lockdown.c:742 lockdownd_client_new_with_handshake(): Session opening failed.

and

11:38:51 lockdown.c:272 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
ERROR: Could not connect to lockdownd: SSL error (-5)

Full output below ... with some 

% ideviceinfo -d      
         
11:38:51 lockdown.c:651 lockdownd_client_new(): device udid: 1c5f8cb1fa98c132c77c29d320aec0be5f5ea2e4
11:38:51 lockdown.c:380 lockdownd_query_type(): called
11:38:51 property_list_service.c:132 internal_plist_send(): sending 284 bytes
11:38:51 service.c:144 service_send(): sending 4 bytes
11:38:51 idevice.c:600 idevice_connection_send(): internal_connection_send 4, sent 4
11:38:51 service.c:144 service_send(): sending 284 bytes
11:38:51 idevice.c:600 idevice_connection_send(): internal_connection_send 284, sent 284
11:38:51 property_list_service.c:137 internal_plist_send(): sent 284 bytes
11:38:51 property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
11:38:51 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
11:38:51 property_list_service.c:211 internal_plist_receive_timeout(): 297 bytes following
11:38:51 property_list_service.c:224 internal_plist_receive_timeout(): received 297 bytes
11:38:51 property_list_service.c:253 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
11:38:51 lockdown.c:396 lockdownd_query_type(): success with type com.apple.mobile.lockdown
11:38:51 property_list_service.c:132 internal_plist_send(): sending 332 bytes
11:38:51 service.c:144 service_send(): sending 4 bytes
11:38:51 idevice.c:600 idevice_connection_send(): internal_connection_send 4, sent 4
11:38:51 service.c:144 service_send(): sending 332 bytes
11:38:51 idevice.c:600 idevice_connection_send(): internal_connection_send 332, sent 332
11:38:51 property_list_service.c:137 internal_plist_send(): sent 332 bytes
11:38:51 property_list_service.c:138 internal_plist_send(): printing 332 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
11:38:51 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
[... 128 lines cut ...]
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>StopSession</string>
	<key>SessionID</key>
	<string>F1769FBD-92EC-477C-AF86-8E38458F4B79</string>
</dict>
</plist>
11:38:51 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
11:38:51 property_list_service.c:211 internal_plist_receive_timeout(): 369295618 bytes following
11:38:51 property_list_service.c:224 internal_plist_receive_timeout(): received 604 bytes
11:38:51 property_list_service.c:224 internal_plist_receive_timeout(): received 7 bytes
11:38:51 idevice.c:642 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -54 (Connection reset by peer)
11:38:51 service.c:167 service_receive_with_timeout(): could not read data
11:38:51 property_list_service.c:229 internal_plist_receive_timeout(): received incomplete packet (611 of 369295618 bytes)
11:38:51 property_list_service.c:231 internal_plist_receive_timeout(): incomplete packet following:
0000: 44 0b 00 02 40 00 02 3d 00 02 3a 30 82 02 36 30   | D...@..=..:0..60
<snip, snip>
0260: 02 01 00                                          | ...

11:38:51 lockdown.c:272 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
ERROR: Could not connect to lockdownd: SSL error (-5)

mexmer, 2024-09-06:
from the error it looks like there is problem with cipher suite, so client tries to renegotiate encryption, but this is disabled (there was bug in this function discovered some years ago).

you might try solution from here
https://github.com/openssl/openssl/issues/21296#issuecomment-1810043699

other than that, you would need to recompile libimobile to force only tls 1.2 when communicating with your device.

cnsxxx09, 2024-09-06:
bloody hell, that solution with the open ssl conf file worked!!!
many many thanks!