libimobiledevice/libimobiledevice #511: device storage extra details

daniel-bluesea, 2017-06-06:
[image]

iTunes display storage divided into sections: apps, books, media, free etc..
it gets the information from iTunes ATH.exe (using pipes)
the plist that has this extra info is called "AssetMetrics" (see attached)

can we get this storage extra info using libimobiledevice?

[assetmetrics.txt](https://github.com/libimobiledevice/libimobiledevice/files/1054919/assetmetrics.txt)

mexmer, 2017-06-07:
that depends how this plist is created.

if it's response from some service, like installation proxy, it will be possible, but if this is some computed value (like ath.exe asks multiple services on iOS device for informations), then you will need to do such stuff yourself, basically you will need ask same servives for related stuff (asset statistics) and compute result from that.

for example, applications themselves are not that hard, you can just ask installation proxy for disk usage
(StaticDiscUsage = application itself, DynamicDiscUsage = application data) if you sum that, you get space occupied by applications.

FreeSpace is easily queried from storage info.

but for the rest ... it's not that easy.

nikias, 2017-06-07:
It used to be possible to derive the storage usage via lockdown:
```ideviceinfo -q com.apple.disk_usage```
They had separate entries for CameraUsage and MobileApplicationUsage but these have disappeared. Now there's an entry ```NANDInfo``` which might contain info on that. Not sure about the format.

mexmer, 2017-06-07:
CameraUsage  and MobileApplicationUsage disappeared like in version 6 or so. (will need to recheck, but they gone for long time already).

atm. for applications i use installation proxy
for ibooks i just calculate size of files in ibooks folders, same for other media (since music and video files are in itunes_control folder, and photos in dcim and photodata)

calculating size by traversing filesystem is quite time consuming tho'

daniel-bluesea, 2017-06-08:
@mexmer thanks for the info. indeed i'm able to extract "AssetMetrics" with small python app (using pipes with ath.exe) i just hoped there is an elegant way using libimobiledevice.

@nikias i'm familiar with ideviceinfo -q com.apple.disk_usage, but it gives general information + NANDInfo that require research in order to parse it.

thanks guys!

daniel-bluesea, 2017-06-11:
@mexmer @nikias btw, when using ideviceinstaller -l the output includes only the first three "headers": CFBundleIdentifier, CFBundleVersion, CFBundleDisplayName and missing the following 2 "headers": StaticDiskUsage & DynamicDiskUsage

from one hand "instproxy_client_options_set_return_attributes" function includes those 2 missing headers, but from the other hand "print_apps_header" function doesn't print out those 2 missing headers.

please advise

daniel-bluesea, 2017-06-11:
@nikias i updated the "print_apps" function to display these values also.
i think you can close this issue

closed 2017-06-12 by daniel-bluesea

AiXanadu, 2024-01-09:
@daniel-bluesea Can you please tell me how to request the "AssetMetrics" you are talking about?