libimobiledevice/libimobiledevice #513: how can i speed idevicescreenshot on large screen device

lujun-wuhan, 2017-06-07:
i need to speed idevicescreenshot on iphone 7 plus or iphone 6s plus，how i can do？

ketan077, 2017-06-16:
com.apple.cmio.iOSScreenCaptureAssistant

pyraman, 2017-06-26:
let's create 10 threads and in each thread we create one screenshotr_client_t.
now we have 10 clients talking to screenshot service.
just an idea - let's try :)

nikias, 2017-06-26:
Multiple threads won't work because IIRC the service doesn't allow to be started more than once.

pyraman, 2017-06-26:
I faced the same problem when i tried to capture my ipad's screenshot. It took about one second to receive a tiff image. This fps was not acceptable for me because i was trying to create something like screencast. I got an idea to create multiple screenshot client.
I'll make POC tomorrow! Will let you know soon.

ketan077, 2017-06-26:
is it possible video data capture from ios devices??

crummy, 2017-06-29:
iMovie does it, for what it's worth.

ketan077, 2017-06-30:
but it work for one device at a time . i want to achieve it for multiple devices at a time using USB lightning cable

xdeng, 2017-08-10:
airplay

xdeng, 2017-08-10:
screenshot   very inefficient

lujun-wuhan, 2017-10-18:
@pyraman Are there any progress?

pyraman, 2017-10-19:
hi wuhan!
sorry for not update! I've been very busy with the release!
How about your side? Is this feature important for you now?

411506656, 2018-01-18:
How to usb  libimobiledevice achieve iphone vote on linux screen？

qmfrederik, 2018-09-04:
Appium now has a fairly efficient screen capture process that runs on top of Xcode UI Test, see https://github.com/appium/WebDriverAgent/pull/105 . You'll still need an Instruments client.

lujun-wuhan, 2018-09-22:
> Appium now has a fairly efficient screen capture process that runs on top of Xcode UI Test, see [appium/WebDriverAgent#105](https://github.com/appium/WebDriverAgent/pull/105) . You'll still need an Instruments client.

Thank you very much!

closed 2019-09-17 by lujun-wuhan