libimobiledevice/libimobiledevice #515: MDM detect.

tomy765, 2017-06-08:
Hello All,

I have a small problem.
How can I check if the device has mdm?
Any ideas.

Thanks,
Tomy

Kanong017, 2017-06-10:
mdm is what?

tomy765, 2017-06-12:
MDM is Mobile Device Management

rcmpayne, 2017-06-12:
i use these tools everyday for mdm and no real way to tell. however, if your mdm Server is pushing a provisioning profile, you can scan for that.

```
echo.
cls
echo **********************************************************
echo ******** Pulling Installed Provisioning Profiles  ********
echo **********************************************************
echo.
mkdir %DIR1%\%DIR16%
echo List profiles
%Tools%\ideviceprovision.exe list > %DIR1%\%FILENAME133%
echo copy profiles
%Tools%\ideviceprovision.exe copy %DIR1%\%DIR16%
```

mexmer, 2017-06-12:
keep in mind, that any iOS device that has FindMyiPhone/FindMyiPad enabled, is subject to MDM. and apple instructs users by default to enable it.

anyways, MDM can be forced either trough provisioning profile (there are tools, that allows you to list provisioning profiles, and of course you need to parse them afterwards). but device can be also subject to MDM when connected to Exchange server trough ActiveSync.

i suspect, there are other ways to enable MDM on device.

nikias, 2017-06-18:
@mexmer he is not asking how to enable it but how to detect if the device is enrolled into an MDM.
if the device is supervised, you can detect it via lockdown:
```
ideviceinfo -q com.apple.mobile.chaperone -k DeviceIsChaperoned
```

mexmer, 2017-06-19:
i know, i was just saying, that nowaydays almost every ios device is subject to mdm, one way or other

NoahReyson, 2019-10-17:
> @mexmer he is not asking how to enable it but how to detect if the device is enrolled into an MDM.
> 
> If the device is supervised, you can detect it via lockdown:
> 
> ```
> ideviceinfo -q com.apple.mobile.chaperone -k DeviceIsChaperoned
> ```
> 
> It will return true or false, respectively.

What are some other keys for this domain pls?
Or where did you find it ?

The --help says default is 'All keys' but it returns nothing for me.

XY-Wing, 2020-11-26:
@nikias @NoahReyson I have the same question that what the other keys for this domain. Do you find a way to get the keys?

tudorcraciun, 2023-03-27:
Hello, I have the same question. Any documentation to find the rest of the keys for **com.apple.mobile.chaperone**? @nikias @NoahReyson @XY-Wing

yunhailu, 2024-05-24:
> Hello, I have the same question. Any documentation to find the rest of the keys for **com.apple.mobile.chaperone**? @nikias @NoahReyson @XY-Wing

@tudorcraciun Do you resolve it？All keys？

tudorcraciun, 2024-05-28:
Hi @yunhailu, I have not made any progress in that direction. Do you have any additional information?