libimobiledevice/libimobiledevice #516: macOS High Sierra - lockdownd_client_new_with_handshake crashes during ssl enable due to wrong dylib

EliyahuStern, 2017-06-11:
Our java process uses libimobiledevice.dylib as native library.

After upgrading to macOS Sierra macOS **High** Sierra... the process crashes with SIGSEGV (0xb) in the following frame:
[libboringssl.dylib+0x7ba20]  ssl_create_cipher_list+0x69
[libboringssl.dylib+0x7cb08]  SSL_CTX_new+0x108

It seems like the issue is the new library at /usr/liblibboringssl.dylib. A temporary workaround we are using is to move this library aside.

nikias, 2017-06-18:
Hmm weird, and that even though it is not linked against libboringssl? What does otool -L libimobildevice.dylib show?

EliyahuStern, 2017-06-18:
I don't have access to a High Sierra machine right now (it had many other issues and I had to re-install Sierra...), but the same library on a Sierra system shows:
$ otool -L libimobildevice.dylib
...
/usr/lib/libssl.0.9.8.dylib (compatibility version 0.9.8, current version 0.9.8)
...

If I'm not mistaken, I had libssl.35 deeper in the stack trace. 

I will give more details once I have the issue reproduced again.

nikias, 2017-07-04:
@EliyahuStern any update on this?

EliyahuStern, 2017-07-05:
I've done `brew install libimobiledevice` again on a different (cleaner) High Sierra machine and everything looks fine.

I guess it was an issue with the machine setup, you can close this one. I'll let you know if I find the cause...

samrocketman, 2017-07-05:
You can close it yourself because you opened it.

EliyahuStern, 2017-07-05:
Ok thanks

closed 2017-07-05 by EliyahuStern