libimobiledevice/libimobiledevice #517: iOS 10.3.1, iPhone 6, backup works, unback doesn't

jpvlsmv, 2017-06-16:
Using yesterday's master/ of all components, I am able to connect my iPhone 6, and read the identifier, pair successfully, read syslog, and read info just fine.  idevicebackup2 backup runs to completion, and I get ~2.5GB of unencrypted data written into the backup directory.

idevicebackup2 unback, info, and list fail.

```
ubuntu@ubuntu-xenial:~$ idevicebackup2 unback ~/backup/
Backup directory is "/home/ubuntu/backup/"
Started "com.apple.mobilebackup2" service on port 49339.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting to unpack backup...
Backup password: No
Sending 'e6e9d57fec7483f45e61c3c3fc01572fd4a02f1e/Manifest.plist' (31.0 KB)
Sending 'e6e9d57fec7483f45e61c3c3fc01572fd4a02f1e/Manifest.db' (11.5 MB)
Device is not ready yet. Going to try again in 2 seconds...
Device is not ready yet. Going to try again in 2 seconds...
Device is not ready yet. Going to try again in 2 seconds...
^CExiting...
Unback Aborted.
```
I would guess the error may be in the syslog output:

There's a missing entitlement, and a crash in BackupAgent2.
```
Jun 16 13:57:58 Alexandrinas-iPhone BackupAgent2[311] <Notice>: Action: Unback
Jun 16 13:57:58 Alexandrinas-iPhone BackupAgent2[311] <Notice>: Backup properties: <private>
Jun 16 13:58:00 Alexandrinas-iPhone backupd[269] <Notice>: Client connection from '<private>' missing entitlement backupd-connection-initiate
Jun 16 13:58:00 Alexandrinas-iPhone backupd[269] <Notice>: Client connection from '<private>' missing entitlement backupd-connection-initiate
Jun 16 13:58:00 Alexandrinas-iPhone BackupAgent2(CoreFoundation)[311] <Notice>: -[MBManifestDB fileEnumerator]: unrecognized selector sent to instance 0x100455950
Jun 16 13:58:00 Alexandrinas-iPhone BackupAgent2(CoreFoundation)[311] <Notice>: *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[MBManifestDB fileEnumerator]: unrecognized selector sent to instance 0x100455950'
*** First throw call stack:
(0x18b456fd8 0x189eb8538 0x18b45def4 0x18b45af4c 0x18b356d2c 0x1000f081c 0x1000c58d8 0x1000c47f4 0x1980cf618 0x1980d0698 0x18b3ef780 0x18b405adc 0x18b40527c 0x18b402d90 0x18b332d94 0x18b37f9a0 0x1000c39bc 0x1000c2ff8 0x18a34159c)
Jun 16 13:58:00 Alexandrinas-iPhone ReportCrash(CrashReporterSupport)[312] <Notice>: cr_update: <private>
```

[syslog.txt](https://github.com/libimobiledevice/libimobiledevice/files/1081638/syslog.txt)
[unback.txt](https://github.com/libimobiledevice/libimobiledevice/files/1081637/unback.txt)
[usbmuxd.txt](https://github.com/libimobiledevice/libimobiledevice/files/1081639/usbmuxd.txt)

nikias, 2017-06-18:
Apple decided to break this. Not sure if on purpose or by accident. This is not a bug in libimobiledevice and nothing that can be fixed from our side...

closed 2017-06-18 by nikias

daniel-bluesea, 2017-11-28:
i think you dont need the unback functionality anymore, cause apple creates a Manifest.db file (during backup) which is simple sqlite file and there you have your files list with there names and relative path