libimobiledevice/libimobiledevice #518: Circular reference error with libusbmuxd while building

fwd079, 2017-06-17:
Hi, I downloaded git repositories for ifuse, libusbmuxd and libimobiledevice.
ifuse built successfully.

But for both libusbmuxd and libimobiledevice, the autogen.sh fails, complaining the other bit is not there. So you cannot run autogen.sh for either libusbmuxd or libimobiledevice.

Please can someone guide me how to build these? Thanks.

nikias, 2017-06-17:
No need to double post here and on the usbmuxd project.
Your problem is that you confuse usbmuxd with libusbmuxd. libusbmuxd does not require libimobiledevice but usbmuxd does. Hence build in the following order:
- libplist
- libusbmuxd
- libimobiledevice
- usbmuxd

fwd079, 2017-06-19:
Okay, thanks. I will try it and let you know.

fwd079, 2017-06-24:
Thanks @nikias  all is installed successfully, now I'll go and check what to do next.

closed 2017-06-24 by fwd079