libimobiledevice/libimobiledevice #519: iOS 11 idevicebackup2 unback issues

rcmpayne, 2017-06-18:
The unback starts and pulls some data but stops after a few logging "Device is not ready yet. Going to try again in 2 seconds..." here is a -d unback  compiled from msys2

```
12:09:39 property_list_service.c:128 internal_plist_send(): sending 106 bytes
12:09:39 service.c:140 service_send(): sending 4 bytes
12:09:39 service.c:140 service_send(): sending 106 bytes
12:09:39 property_list_service.c:133 internal_plist_send(): sent 106 bytes
12:09:39 property_list_service.c:134 internal_plist_send(): printing 344 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
        <string>DLMessageStatusResponse</string>
        <integer>0</integer>
        <string>___EmptyParameterString___</string>
        <string>___EmptyParameterString___</string>
</array>
</plist>
12:09:39 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:39 service.c:163 service_receive_with_timeout(): could not read data
12:09:39 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:39 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
12:09:40 notification_proxy.c:48 np_lock(): Locked
12:09:40 service.c:163 service_receive_with_timeout(): could not read data
12:09:40 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:40 notification_proxy.c:59 np_unlock(): Unlocked
12:09:41 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:41 service.c:163 service_receive_with_timeout(): could not read data
12:09:41 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:41 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
12:09:41 notification_proxy.c:48 np_lock(): Locked
12:09:42 service.c:163 service_receive_with_timeout(): could not read data
12:09:42 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:42 notification_proxy.c:59 np_unlock(): Unlocked
12:09:43 notification_proxy.c:48 np_lock(): Locked
12:09:43 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:43 service.c:163 service_receive_with_timeout(): could not read data
12:09:43 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:43 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
12:09:43 service.c:163 service_receive_with_timeout(): could not read data
12:09:43 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:43 notification_proxy.c:59 np_unlock(): Unlocked
12:09:44 notification_proxy.c:48 np_lock(): Locked
12:09:45 service.c:163 service_receive_with_timeout(): could not read data
12:09:45 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:45 notification_proxy.c:59 np_unlock(): Unlocked
12:09:45 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:45 service.c:163 service_receive_with_timeout(): could not read data
12:09:45 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:45 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
12:09:46 notification_proxy.c:48 np_lock(): Locked
12:09:46 service.c:163 service_receive_with_timeout(): could not read data
12:09:46 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:46 notification_proxy.c:59 np_unlock(): Unlocked
12:09:47 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:47 service.c:163 service_receive_with_timeout(): could not read data
12:09:47 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:47 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
12:09:47 notification_proxy.c:48 np_lock(): Locked
12:09:48 service.c:163 service_receive_with_timeout(): could not read data
12:09:48 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:48 notification_proxy.c:59 np_unlock(): Unlocked
12:09:49 notification_proxy.c:48 np_lock(): Locked
12:09:49 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:49 service.c:163 service_receive_with_timeout(): could not read data
12:09:49 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:49 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
12:09:49 service.c:163 service_receive_with_timeout(): could not read data
12:09:49 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:49 notification_proxy.c:59 np_unlock(): Unlocked
12:09:50 notification_proxy.c:48 np_lock(): Locked
12:09:51 service.c:163 service_receive_with_timeout(): could not read data
12:09:51 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
12:09:51 notification_proxy.c:59 np_unlock(): Unlocked
12:09:51 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (File exists)
12:09:51 service.c:163 service_receive_with_timeout(): could not read data
12:09:51 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
12:09:51 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
Exiting...
Unback Aborted.
12:09:52 property_list_service.c:128 internal_plist_send(): sending 97 bytes
12:09:52 service.c:140 service_send(): sending 4 bytes
12:09:52 service.c:140 service_send(): sending 97 bytes
12:09:52 idevice.c:352 internal_connection_send(): ERROR: usbmuxd_send returned -32 (Broken pipe)
12:09:52 service.c:143 service_send(): ERROR: sending to device failed.
12:09:52 property_list_service.c:143 internal_plist_send(): ERROR: sending to device failed.
12:09:52 property_list_service.c:128 internal_plist_send(): sending 235 bytes
12:09:52 notification_proxy.c:48 np_lock(): Locked
12:09:52 service.c:140 service_send(): sending 4 bytes
12:09:52 service.c:140 service_send(): sending 235 bytes
12:09:52 property_list_service.c:133 internal_plist_send(): sent 235 bytes
12:09:52 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
12:09:52 property_list_service.c:134 internal_plist_send(): printing 235 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>Shutdown</string>
</dict>
</plist>
12:09:52 property_list_service.c:202 internal_plist_receive_timeout(): 237 bytes following
12:09:52 notification_proxy.c:132 np_client_free(): joining np callback
12:09:52 property_list_service.c:215 internal_plist_receive_timeout(): received 237 bytes
12:09:52 property_list_service.c:241 internal_plist_receive_timeout(): printing 237 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>ProxyDeath</string>
</dict>
</plist>
12:09:52 notification_proxy.c:289 np_get_notification(): NotificationProxy died!
12:09:52 notification_proxy.c:59 np_unlock(): Unlocked
```

nikias, 2017-06-18:
See https://github.com/libimobiledevice/libimobiledevice/issues/517

rcmpayne, 2017-06-19:
close as dup. no intent to fix

closed 2017-06-19 by rcmpayne